78/xiaozhi-esp32 #563: 希望可以后台添加音乐源

woshichenghaibo, 2025-05-01:
### Is your feature request related to a problem?

绝大部分音乐都不可以播放。我就是播放个射雕英雄传主题曲都没有版权？问题是这玩意儿不需要版权啊

### Describe the solution you'd like.

可以在后台设置音乐源，然后播放自己想听的音乐

78, 2025-05-01:
自行搭建开源server没有这个限制

woshichenghaibo, 2025-05-05:
有教程吗？

woshichenghaibo, 2025-05-05:
> 自行搭建开源server没有这个限制
就算如此，可是消费者并不是人人一台8GB显卡的服务器呀，能否提供webdav接口进去，我自己在nat服务器上搭建的alist服务里面有很多音乐，可以调用搜索播放印方，这样就会好很多吧

maobaolong, 2025-05-05:
可以设置百度网盘，夸克网盘

78, 2025-05-06:
> 有教程吗？
> […](#)

B站和抖音很多吧

woshichenghaibo, 2025-05-09:
找到了很多搭建小智服务器的，是ai服务器不是音乐服务器，ai服务器需要8GB显存，还得长期开机。。。

jiangyanfeng, 2025-05-11:
> > 有教程吗？
> > […](#)
> 
> B站和抖音很多吧

希望小智服务器上能让客户绑定QQ音乐账户、喜马拉雅账户、蜻蜓FM账户等，然后播放里面的音乐。
这样通过音乐内容源官方API接入，也不会有版权问题了。

jiangyanfeng, 2025-05-11:
除了版权原因，还有一个原因就是会造成服务器压力大。
建议虾哥搞个收费会员，提供上述功能。

我目前在实现 TF 卡本地音乐/在线FM广播播放功能，不需要服务器支持。
源码混合了 ESP-ADF，没有采用用双固件分区的方式，故可以通过小智语音控制播放（本地音乐播放、上一首、下一首、停止、暂停、顺序/随机/单曲循环播放模式）。
硬件上需要板子带 TF 卡，Mic 与 Speaker 分腔隔离，带硬件回采更好。

jiangyanfeng, 2025-05-16:
我目前的做法和你类似，只是不自己搭建音乐源服务器，使用网易云音乐、QQ音乐的API接入，播放上面的音乐。从网易云音乐、QQ音乐的开发者文档上看是可以实现的。
目前已通过ioT模式实现播放本地 TF 卡上的音乐，并可搜索设备本地TF卡歌曲名来播放。

maobaolong, 2025-05-17:
可否播放百度网盘里的内容，如果这样，就可以语音点播故事 mp3听了

woshichenghaibo, 2025-05-17:
具体如何实现的请分享一下？

jiangyanfeng, 2025-05-17:
需要 http_stream，目前BUG较多。建议等虾哥MCP功能上了后再搞。

woshichenghaibo, 2025-05-20:
> 需要 http_stream，目前BUG较多。建议等虾哥MCP功能上了后再搞。

https://github.com/hanxi/xiaomusic

这位大侠已经在小爱音箱实现了。
About
使用小爱音箱播放音乐，音乐使用 yt-dlp 下载。

[xdocs.hanxi.cc/](http://xdocs.hanxi.cc/)

首先播放docker本地音乐，如果本地不存在则先去下载再播放，全程自动