78/xiaozhi-esp32 #574: 增加人体感应功能

yezhihao312, 2025-05-05:
### Is your feature request related to a problem?

小智设备还没到手之前，是因为看到米思齐的模块式编程，以为可以简单的推动模块实现功能。结果是否的。所以现在有个想法，增加人体感应功能，可以让设备更智能。

### Describe the solution you'd like.

刚开始可以简单点，能检测到人体接近会自动触发问候功能就行，为后面大佬们的产品方向起了个头。如：在外接电源的情况下，在几个小时后触发一次（如：你回来啦），在一定时间检测到没有活动，会提醒喝水、走动。抖音搜了下资料发现像卖600元的任天堂闹钟alarmo，用的毫米波雷达检测任务姿态来检测是否起床、赖床，然后播放对应的音乐。设备要求，低功耗，体积小。模块单价在DIY爱好者能接受的范围内，几十块。

市面上的产品有老人跌倒检测，这个是放在屋顶的，这不是想要的。我想要的是类似人体感应灯那样，接近迟发对话功能，但是红外的感应器功能没有毫米波雷达好用。后期可以延伸到，姿态管理，如：小智在屏幕里面跟着你做动作，或者感应到人的动作超出了一定值后会发出提醒（礼仪培训班，地铁工作人员的规范手势），哈哈，这个说的太大了。

### Describe alternatives you've considered.

目前在研究者用哪种毫米波雷达合适，要求：低功耗，体积小，精度合适，性价比高。了解好后，后续会给出建议。

yezhihao312, 2025-05-05:
补充一下：为什么老是要我主动去跟设备聊天，设备就不能主动一点吗，主动点会更加的有爱，增加与机器交互的频率，让人与机器更加有粘性。

yezhihao312, 2025-05-05:
抖音看到一位大佬发布的零成本加WiFi雷达的视频。网上搜了下，https://blog.csdn.net/weixin_47877869/article/details/146313346，这篇文章提到了Wi‑Fi CSI（Channel state information）技术及其具体示例解析。也提到了ESP32 CSI 示例。可以考虑在不增加硬件的情况下添加这项功能。

如果外接硬件的话，我看这款超低功耗的人体感应雷达不错“HLK-LD2410S”。

yezhihao312, 2025-05-05:
抖音看到一位大佬发布的零成本加WiFi雷达的视频。网上搜了下，https://blog.csdn.net/weixin_47877869/article/details/146313346 
这篇文章提到了Wi‑Fi CSI（Channel state information）技术及其具体示例解析。也提到了ESP32 CSI 示例。可以考虑在不增加硬件的情况下添加这项功能。

如果外接硬件的话，我看这款超低功耗的人体感应雷达不错“HLK-LD2410S”。

yezhihao312, 2025-05-05:
4.64 复制打开抖音，看看【启明云端科技的作品】启明云端乐鑫代理商，乐鑫ESP-CSI智能人体感知... https://v.douyin.com/TbS9XvfvNdw/ 07/01 eOX:/ W@M.Wz

看了这个视频介绍，是可以实现功能的。还能检测活动次数（太恐怖了，呵呵，老王捕捉器）

layola13, 2025-05-06:
补充一下：为什么老是要我主动去跟设备聊天，设备就不能主动一点吗，主动点会更加的有爱，增加与机器交互的频率，让人与机器更加有粘性。

有些人不喜欢骚扰，其实你可以考虑用自定义后台配置这样的特性，

bobjeey, 2025-05-07:
这个想法太好了，特别适合无人自助的商业项目

closed 2025-05-20 by 78