78/xiaozhi-esp32 #582: 小智二开AddStringProperty和AddMethod会莫名奇妙卡死

lovelyyoshino, 2025-05-07:
我是在IOT中加入了一个类，用于让小智和其他传感器进行交互，当中涉及串口的接收和发送，但是出现了下面的现象

[image]

正常的现象如下
[image]

通过一行注释发现代码只需要加入这一行就会有问题

[image]

相关代码如下

```cpp
        // Add properties for monitoring robot status
        properties_.AddBooleanProperty("is_navigating", "是否在导航中", [this]() -> bool {
            return is_navigating_;
        });
        
        properties_.AddStringProperty("current_destination", "当前目的地", [this]() -> std::string {
            return current_destination_;
        });
        
        properties_.AddStringProperty("detected_object", "检测到的物体", [this]() -> std::string {
            return detected_object_;
        });
        
        properties_.AddBooleanProperty("emergency_stop", "紧急停止状态", [this]() -> bool {
            return emergency_stop_;
        });

        properties_.AddStringProperty("navigation_status", "导航状态", [this]() -> std::string {
            return navigation_status_;
        });
        
        // Define methods for controlling the robot
        methods_.AddMethod("NavigateTo", "导航到指定地点", 
            ParameterList(std::vector<Parameter>{
                Parameter("destination", "目的地名称", kValueTypeString, true)
            }), [this](const ParameterList& parameters) {
            std::string destination = static_cast<std::string>(parameters["destination"].string());
            current_destination_ = destination;
            is_navigating_ = true;
            emergency_stop_ = false;
            
            // 发送导航命令
            SendUartData(CMD_NAVIGATE, destination.c_str());
            
            // 语音确认
            char voice_content[200];
            snprintf(voice_content, sizeof(voice_content) - 1, "开始导航到%s", destination.c_str());
            SendVoiceMessage(voice_content);
        });
        
        // methods_.AddMethod("Stop", "停止导航", 
        //     ParameterList(), [this](const ParameterList& parameters) {
        //     is_navigating_ = false;
        //     SendUartData(CMD_STOP, "");
            
        //     // 语音确认
        //     SendVoiceMessage("已停止导航");
        // });
        
```

lovelyyoshino, 2025-05-07:
加入properties_.AddStringProperty("navigation_status", "导航状态", [this]() -> std::string {
            return navigation_status_;
        });和methods_.AddMethod("Stop", "停止导航", 都会造成上面的现象，我看代码层面是没问题的，不知道是不是小智内部什么机制
@78

lovelyyoshino, 2025-05-08:
@78 虾哥可以看一下这个问题嘛

78, 2025-05-08:
> 加入properties_.AddStringProperty("navigation_status", "导航状态", this -> std::string { return navigation_status_; });和methods_.AddMethod("Stop", "停止导航", 都会造成上面的现象，我看代码层面是没问题的，不知道是不是小智内部什么机制 [@78](https://github.com/78)

去掉其他的，只保留这个看看。检查是否内容太多导致无法发送成功。

lovelyyoshino, 2025-05-09:
这个是有什么限制嘛，这个是否可以拓展？

78, 2025-05-09:
> 这个是有什么限制嘛，这个是否可以拓展？

如果用了ml307的uart通信，对包大小有限制

78, 2025-05-09:
> 我用的是正点的板子，我删掉只留一个确实可以，这个问题可以怎么解决呢

重新测试一下看看，调整了服务器这边的参数

lovelyyoshino, 2025-05-10:
解决了，谢谢虾哥

closed 2025-05-10 by lovelyyoshino