78/xiaozhi-esp32 #584: Receiving White noise in speaker

hs0747, 2025-05-08:
Hi, First of all thanks for this source. I am super happy with your repository.
My issue is, I have used ESP32-S3 N16R8 Development board. I have chosen the bread-compact-wifi with 128x64 OLED display. I have compiled and flash to the ESP32S3 board. After flashed i am getting White Noise in Speaker. I tried different method still receiving noise. 

I compiled and flashed using VS-Code. I have tried with ESP-idf v5.4.1, v5.3.3 and v5.3.1.

78, 2025-05-08:
It is possible that there is an issue with the wire connection.

Have you followed this guide to assemble your board?!

[image]

hs0747, 2025-05-08:
Thank you, i did connection as per above. 
No noise in v1.6.0 firmware. 
now i receiving noise in v1.6.2. 
Also VS code complied build.

78, 2025-05-08:
Could you kindly provide a video or audio recording to demonstrate the noise you have referred to?

hs0747, 2025-05-09:
i tried to compile the latest v1.6.2 with esp-idf v5.3.1, it's working perfectly. 

https://github.com/user-attachments/assets/0705630f-3f7d-4236-814a-425a3bc5bd33

but if i tried to flash your .bin file, noise coming.

https://github.com/user-attachments/assets/94e0935f-2153-43d6-8990-0783e63c25e9

note:
 even if i change the esp-idf v5.3.1 to latest version, noise is coming..

78, 2025-05-09:
I have tested the v1.6.2 firmware and have not encountered the problem you mentioned. Considering that I heard a screeching sound during the initial stage of the booting process, I suspect that there is still an issue with the wire connections on your development board. It is advisable to conduct a general inspection of the GND wire connected to the amplifier.

hs0747, 2025-05-09:
Have you seen my second video, that i have mentioned white noise. it was not in previous version 1.6.0.
anyhow once again i verify my circuit. 
thanks for your quick response. :)

hs0747, 2025-05-09:
Issue solved. there is an internal short between gpio 6 and 7. Now working perfectly. 
Thank you. :)