78/xiaozhi-esp32 #585: 1.6.2无法连接自部署服务器ota返回的ws地址

AiBoBoMaker, 2025-05-08:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.6.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

希望能够正常连接服务器

### What is the actual behavior?

我是1.6.2版本，从自己搭建的[xinnan老哥的项目后，从设备的日志里看到，ota获取到了正确的ws地址，但是设备无法连接；我换了1.5.6版本，直接写这个ws地址是可以连接的。
是我哪里配置不对？

### Steps to reproduce.

我是1.6.2版本，从自己搭建的[xinnan老哥的项目后，从设备的日志里看到，ota获取到了正确的ws地址，但是设备无法连接；我换了1.5.6版本，直接写这个ws地址是可以连接的。
是我哪里配置不对？

### Debug Logs.

```plain
I (383542) Application: STATE: connecting
I (383552) WS: Connecting to websocket server: ws://192.168.1:4:8015/xiaozhi/v1/ with version: 1
I (383552) WebSocket: Connecting to ws://192.168.1:4:8015/xiaozhi/v1/
I (393242) Application: Free internal: 21107 minimal internal: 19939
E (402062) TcpTransport: Failed to connect to 192.168.1:4
E (402062) WebSocket: Failed to connect to server
E (402062) WS: Failed to connect to websocket server
I (402062) Application: STATE: idle
W (402072) Application: Alert 错误: 正在寻找可用服务 [sad]
I (402102) AudioCodec: Set output enable to true
I (411242) Application: Free internal: 20191 minimal internal: 19939
```

78, 2025-05-08:
你需要更新服务器版本。

openrz, 2025-05-09:
配置文件里，ip地址配错了好像。不应该是192.168.1:4，而是192.168.1.4

AiBoBoMaker, 2025-05-09:
还真的是。。。可能是昨天睡太迟了，眼花缭乱了哈哈

closed 2025-05-20 by 78