78/xiaozhi-esp32 #587: 最新1.6.2版4G面包板128x64固件，双击boot按钮无法切换网络

manicwy, 2025-05-08:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

1.6.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Power Supply used.

Battery

### What is the expected behavior?

能够正常实现4G网络和Wifi网络的连接切换

### What is the actual behavior?

双击boot按钮时，当第一次按键按下去就立即响应单击重启的动作了，在检测4G模组的时候无论怎么双击也无法成功实现网络切换

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

78, 2025-05-08:
检测模组的时候，单击boot是你是单击了RST重启按钮吗

manicwy, 2025-05-08:
> 检测模组的时候，单击boot应该是没有反应的，你是单击了RST重启按钮吗

确实是我理解错误，按照你的提示切换OK，谢谢！

closed 2025-05-20 by 78