78/xiaozhi-esp32 #590: 升级到4月28日之后的版本导致不断重启

chao-from-light-speed, 2025-05-09:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

latest-esp32-s3-touch-lcd-1.46

### Operating System used.

macOS

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

获取OTA信息后不断重启

### What is the actual behavior?

正常启动

### Steps to reproduce.

1. commit id: 5f8389c135f589f9a3b63fe4fd7893f72a097b03 之前的main分支代码可以正常运行，之后都有问题

### Debug Logs.

```plain
I (4473) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (4483) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (4543) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (6273) Backlight: Set brightness to 0
I (7753) wifi: Got IP: 192.168.1.200
I (7753) esp_netif_handlers: sta ip: 192.168.1.200, mask: 255.255.255.0, gw: 192.168.1.1
I (7753) Application: STATE: activating
I (7753) Ota: Current version: 1.6.2
I (7763) EspHttp: Opening HTTP connection to https://api.tenclass.net/xiaozhi/ota/

***ERROR*** A stack overflow in task main has been detected.

Backtrace: 0x4037a031:0x3fcb29c0 0x40384a35:0x3fcb29e0 0x4038592a:0x3fcb2a00 0x403872e7:0x3fcb2a80 0x403859f0:0x3fcb2aa0 0x403859e6:0x00000000 |<-CORRUPTED
--- 0x4037a031: panic_abort at /Users/zhouchao/esp/v5.4/esp-idf/components/esp_system/panic.c:454
0x40384a35: esp_system_abort at /Users/zhouchao/esp/v5.4/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4038592a: vApplicationStackOverflowHook at /Users/zhouchao/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:563
0x403872e7: vTaskSwitchContext at /Users/zhouchao/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3701 (discriminator 7)
0x403859f0: _frxt_dispatch at /Users/zhouchao/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x403859e6: _frxt_int_exit at /Users/zhouchao/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246
```

### More Information.

注释掉ota.cc里的161行 settings.SetInt(item->string, item->valueint); 后可以正常运行

78, 2025-05-09:
git拉一下最新代码看看，我这里测试task main是有4KB的剩余堆栈的，应该不至于发生堆栈溢出。
你可以打印一下 `item->string, item->valueint` 这两个值的内容。

chao-from-light-speed, 2025-05-10:
@galoisgroupcn 并不是空指针的问题。

@78 代码已经是main分支最新代码了。加了输出看了下，见下图，只要打开`settings.SetInt`这个注释就一定会不停重启。换了一块板子也是同样的问题。
[image]

chao-from-light-speed, 2025-05-10:
esp-idf开发我是新手，不太理解为什么`if (item->type == cJSON_Number)`这个条件都不满足，没有进入SetInt这部分也会引起系统不停重启。但注释掉这个部分确实就好了。

78, 2025-05-10:
> esp-idf开发我是新手，不太理解为什么`if (item->type == cJSON_Number)`这个条件都不满足，没有进入SetInt这部分也会引起系统不停重启。但注释掉这个部分确实就好了。

可以加入以下代码来打印堆栈剩余大小：
ESP_LOGI(TAG, "Remaining task stack size: %d", uxTaskGetStackHighWaterMark(NULL));

closed 2025-05-20 by 78