78/xiaozhi-esp32 #596: 新增开发板编译报错/main/boards/common/board.h:30:(.text._ZN5Board11GetInstanceEv[_ZN5Board11GetInstanceEv]+0xe): undefined reference to `create_board()' collect2.exe: error: ld returned 1 exit status ninja: build stopped: subcommand failed.

zxtkj, 2025-05-09:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.6.2 

### Operating System used.

Windows

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

增加一个开发板，zxt-k120c-s3

### What is the actual behavior?

编译后报错如下
FAILED: xiaozhi.elf
cmd.exe /C "cd . && D:\Espressif\5.31\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=E:/Git_KLJ/xiaozhi-esp32-main/build/xiaozhi.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.wdt.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\xiaozhi.elf.rsp -o xiaozhi.elf  && cd ."
D:/Espressif/5.31/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(display.cc.obj):(.literal._ZN5Board11GetInstanceEv[_ZN5Board11GetInstanceEv]+0x8): undefined reference to `create_board()'
D:/Espressif/5.31/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(display.cc.obj): in function `Board::GetInstance()':
E:/Git_KLJ/xiaozhi-esp32-main/main/boards/common/board.h:30:(.text._ZN5Board11GetInstanceEv[_ZN5Board11GetInstanceEv]+0xe): undefined reference to `create_board()'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

zxtkj, 2025-05-09:
使用的是vscode idf 点构建项目，后报的上面的错，选别的开发板是可以正常构建的，就是新增后报错

78, 2025-05-09:
full clean一下项目或者重新 set-target

zxtkj, 2025-05-10:
> full clean一下项目或者重新 set-target

fullclean 和重新 set-target 还是一样的报错，使用idf.py build 就会报这个错，如果使用 python scripts/release.py zxt-k120c-s3 编译就不会报错

zxtkj, 2025-05-10:
使用 python scripts/release.py zxt-k120c-s3 编译就不会报错,但是这样编译出来的不能用，反复重启，输出如下日志：
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403ccbd9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x568
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc14
load:0x403cb700,len:0x2d3c
entry 0x403c8904
ESP-ROM:esp32s3-20210327

zxtkj, 2025-05-11:
fullclean 和重新 set-target 还是一样的报错，使用idf.py build 就会报这个错，如果使用 python scripts/release.py zxt-k120c-s3 编译就不会报错

使用 python scripts/release.py zxt-k120c-s3 编译就不会报错,但是这样编译出来的不能用，反复重启，输出如下日志：
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403ccbd9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x568
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc14
load:0x403cb700,len:0x2d3c
entry 0x403c8904
ESP-ROM:esp32s3-20210327

zxtkj, 2025-05-11:
@78 能帮忙看一下吗？还是没有解决

lostkk2024, 2025-05-13:
mark一下 我也遇到了这个问题

78, 2025-05-26:
> full clean一下项目或者重新 set-target

试试这个方法，不行的话，删除build目录

yxmstar, 2025-07-02:
> 我也遇到这个问题，16.5版本，好像是SDKconfig的问题

请问解决了吗 @kajsdo

SImonWangLK, 2026-02-06:
我也遇到了同样的问题，请问兄弟解决了吗