78/xiaozhi-esp32 #600: 使用OTA升级固件报错

104gogo, 2025-05-11:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

1.6.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

从 OTA 服务下载完之后能正常升级

### What is the actual behavior?

我自己生成的固件包，使用 flash tool 工具能正常烧录和使用
然后使用 OTA 下载完成之后，出现下面报错
```
I (250301) Ota: Progress: 98% (7524735/7661568), Speed: 33580B/s
I (251331) Ota: Progress: 98% (7556755/7661568), Speed: 32020B/s
I (252371) Ota: Progress: 99% (7590335/7661568), Speed: 33580B/s
I (253411) Ota: Progress: 99% (7622555/7661568), Speed: 32220B/s
I (254461) Ota: Progress: 99% (7656135/7661568), Speed: 33580B/s
I (254601) Ota: Progress: 100% (7661568/7661568), Speed: 5433B/s
I (254611) Ml307Http: HTTP连接已关闭，ID: 0
I (254611) esp_image: segment 0: paddr=00880020 vaddr=3fce2820 size=0056ch (  1388)
E (254611) boot_comm: Image requires efuse blk rev >= v251.85, but chip is v1.3
E (254621) Ota: Image validation failed, image is corrupted
I (254631) Application: Firmware upgrade failed...
I (257631) Application: Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037a245
```

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

104gogo, 2025-05-11:
使用虾哥提供的 1.6.0 的固件通过 OTA 升级也会报错同样的错误，我的设备是星智1.54TFT 4g版本
```
I (148981) Ota: Progress: 98% (4451735/4508208), Speed: 33580B/s
I (150001) Ota: Progress: 99% (4483955/4508208), Speed: 32220B/s
I (150721) Ota: Progress: 100% (4508208/4508208), Speed: 24253B/s
I (150741) Ml307Http: HTTP连接已关闭，ID: 0
I (150741) esp_image: segment 0: paddr=00880020 vaddr=3fce2820 size=0056ch (  1388)
E (150741) boot_comm: Image requires efuse blk rev >= v251.85, but chip is v1.3
E (150741) Ota: Image validation failed, image is corrupted
I (150761) Application: Firmware upgrade failed...
I (153761) Application: Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
```

openrz, 2025-05-11:
你上传到自建的 OTA 服务器的bin文件，是xiaozhi.bin，还是合并后的merged-binary.bin？

104gogo, 2025-05-11:
python release.py 运行出来的是 merged-binary.bin，是要改名字吗？有次我改了次试了下好像也不行讷。 @openrz

104gogo, 2025-05-11:
> 你上传到自建的 OTA 服务器的bin文件，是xiaozhi.bin，还是合并后的merged-binary.bin？

合并后的merged-binary.bin

openrz, 2025-05-11:
不能用合并后的merged-binary.bin。在build文件夹，运行python release.py之前，有一个xiaozhi.bin文件，用这个。

merged-binary.bin包含了xiaozhi.bin，你可以理解成，重装的时候用merged-binary.bin，而xiaozhi.bin只是用来升级的。

所以你在服务器里，需要上传的是合并前的xiaozhi.bin

[image]

104gogo, 2025-05-11:
哦哦，感谢，我没理解到它们两个是不一样的

104gogo, 2025-05-11:
好使，感谢 @openrz

closed 2025-05-11 by 104gogo