78/xiaozhi-esp32 #603: 1.6.2版本中MQTT模式下topic是如何被订阅的？

herbvaan, 2025-05-11:
使用mqtt模式时 在main\protocols\mqtt_protocol.cc文件中回调方法OnMessage可以收到消息，但是在代码中并没有找到订阅某个主题的操作，我想问下这个订阅是怎么实现的
[image]

myifeng, 2025-05-16:
应该是在服务器端进行控制的，目前Topic都没有公开

78, 2025-05-16:
是的，阿里云的MQTT Broker与小智目前用的MQTT Broker，都支持给设备发送点对点消息的能力。
小智的 MQTT Broker to Websocket 项目代码也在开源计划中。

myifeng, 2025-05-16:
> 是的，阿里云的MQTT Broker与小智目前用的MQTT Broker，都支持给设备发送点对点消息的能力。
> 小智的 MQTT Broker to Websocket 项目代码也在开源计划中。

虾哥，MQTT协议什么时候可以开源呀

78, 2025-05-16:
> > 是的，阿里云的MQTT Broker与小智目前用的MQTT Broker，都支持给设备发送点对点消息的能力。
> > 小智的 MQTT Broker to Websocket 项目代码也在开源计划中。
> 
> 虾哥，MQTT协议什么时候可以开源呀

https://github.com/78/xiaozhi-mqtt-gateway

closed 2025-05-16 by 78