78/xiaozhi-esp32 #605: 请教下qwen实时模型是如何实现的

PotterSu, 2025-05-12:
### Is your feature request related to a problem?

Hi, 非常不错的工作，感谢开源！有如下几个问题想请教下：
1.想请教下qwen实时模型是基于qwen系列哪个模型实现的
2.体验了小智官方服务器的响应，整个链路的延迟非常低，但是我在机器服务器上搭建了qwen-turbo+豆包的湾湾小何，延迟要1-2s。想请教下有没有什么优化建议。

78, 2025-05-12:
目前Qwen实时主要是 Qwen 2.5 72B。
大模型性能主要看 TTFT 和 TPOT 这两个参数。

PotterSu, 2025-05-12:
好的，感谢大佬回复，我研究下，不懂再请教您

openrz, 2025-05-13:
“实时”相对于“非实时”，在部署优化、连接方式上有什么技巧和区别？

PotterSu, 2025-05-13:
> “实时”相对于“非实时”，在部署优化、连接方式上有什么技巧和区别？

@78 同问。我在楼上大佬开源的xiaozhiserver上部署LLM+tts，响应延迟很高。想请教下：
1.你们的LLM模型是部署在服务器端本地的还是通过阿里api访问的?我通过阿里api访问延迟比较高，同样的qwen2.5-72b，延迟有1-2s。
2.你们也是调用了豆包的弯弯小何api接口吗，我调用了豆包的弯弯小何api，但是延迟有2s左右。还是tts是有其他的部署方式。
感谢！

78, 2025-05-13:
> > “实时”相对于“非实时”，在部署优化、连接方式上有什么技巧和区别？
> 
> [@78](https://github.com/78) 同问。我在楼上大佬开源的xiaozhiserver上部署LLM+tts，响应延迟很高。想请教下： 1.你们的LLM模型是部署在服务器端本地的还是通过阿里api访问的?我通过阿里api访问延迟比较高，同样的qwen2.5-72b，延迟有1-2s。 2.你们也是调用了豆包的弯弯小何api接口吗，我调用了豆包的弯弯小何api，但是延迟有2s左右。还是tts是有其他的部署方式。 感谢！

国内云厂商目前还没有提供实时响应的API，可以考虑私有化部署。TTS使用流式接口。

PotterSu, 2025-05-14:
> > > “实时”相对于“非实时”，在部署优化、连接方式上有什么技巧和区别？
> > 
> > 
> > [@78](https://github.com/78) 同问。我在楼上大佬开源的xiaozhiserver上部署LLM+tts，响应延迟很高。想请教下： 1.你们的LLM模型是部署在服务器端本地的还是通过阿里api访问的?我通过阿里api访问延迟比较高，同样的qwen2.5-72b，延迟有1-2s。 2.你们也是调用了豆包的弯弯小何api接口吗，我调用了豆包的弯弯小何api，但是延迟有2s左右。还是tts是有其他的部署方式。 感谢！
> 
> 国内云厂商目前还没有提供实时响应的API，可以考虑私有化部署。TTS使用流式接口。

好的非常感谢大佬回复！
我之前试了在我们服务器上部署xiaozhi-server项目，访问本地ollama部署的qwen3-4b，延迟仍然很高有500ms-1s。不确定是不是跟我们服务器的算力有关系（RTX5000）。
如果大佬方便的话，可以简单透露下我们这么做是否是对的：
1.我们的服务器是不是算力太低了，应该换个算力高点儿的服务器，比如A100？或者建议的服务器配型。
2.通过ollama的方式部署在本地，然后服务端去访问本地的ollama端口这种方式是否可取。还是最好通过代码的方式直接load模型 推理

78, 2025-05-14:
服务器推理部署应该是用vllm或者sglang，ollama不适合做这些。你需要Hopper架构之后的GPU才能用的上大模型的推理优化。

PotterSu, 2025-05-14:
> 你需要Hopper架构之后的GPU才能用的上大模型的推理优化。

好的，我研究一下，感谢分享！

HiCoderMonkey, 2025-05-14:
@PotterSu 可以尝试使用gpustack部署llm，操作比较简单，支持vllm多机多卡部署

jkwang-a, 2025-06-26:
@PotterSu 老哥，我们这 用户说完话，觉得小智反应了些时间才翻译完成，再发给大模型，再TTS 返回语音。
我觉得 ASR 和 VAD 的延迟很明显，为什么你们搭建的 觉的ASR+VAD 延迟可以忽略了？

我们想搭建一个延迟在1s 左右的，尽可能延迟少的，具体该怎么做：
ASR、？
VAD、？
TTS，用火山的？
LLM 找阿里的 qwen2.5-72b ？

78, 2025-06-26:
> [@PotterSu](https://github.com/PotterSu) 老哥，我们这 用户说完话，觉得小智反应了些时间才翻译完成，再发给大模型，再TTS 返回语音。 我觉得 ASR 和 VAD 的延迟很明显，为什么你们搭建的 觉的ASR+VAD 延迟可以忽略了？
> 
> 我们想搭建一个延迟在1s 左右的，尽可能延迟少的，具体该怎么做： ASR、？ VAD、？ TTS，用火山的？ LLM 找阿里的 qwen2.5-72b ？

如果云厂商没有提供低延迟的大模型API，个人使用的话可以考虑本地部署。72b需要8张3090，qwen3 32b需要4张3090

aeolus21, 2025-07-11:
我这边也自己基于开源server搭了一套，tts也用的豆包的湾湾小何，延时也很大，楼主后来解决了吗

PotterSu, 2025-07-11:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，您的邮件我已收到。我将在短时间内，尽快给您回复。如有急事请拨打18646308557

aeolus21, 2025-07-11:
> 用火山的tts 基本在300ms 很快。llm用 阿里新出的qwen3基础上的 truber-latest模型也很快一般问题llm+tts 在500-700ms左右，现在主要是asr 我用的豆包，还可以吧总的下来1s左右发自我的手机-------- 原始邮件 --------发件人： aeolus21 ***@***.***>日期： 2025年7月11日周五 中午11:31收件人： 78/xiaozhi-esp32 ***@***.***>抄送： trberc ***@***.***>, Comment ***@***.***>主    题： Re: [78/xiaozhi-esp32] 请教下qwen实时模型是如何实现的 (Issue [#605](https://github.com/78/xiaozhi-esp32/issues/605))aeolus21 left a comment ([78/xiaozhi-esp32#605](https://github.com/78/xiaozhi-esp32/issues/605))
> 我这边也自己基于开源server搭了一套，tts也用的豆包的湾湾小何，延时也很大，楼主后来解决了吗

您说的火山tts是HuoshanDoubleStreamTTS 双流式吗

jkwang-a, 2025-07-15:
是的发自我的手机-------- 原始邮件 --------发件人： aeolus21 ***@***.***>日期： 2025年7月11日周五 下午2:56收件人： 78/xiaozhi-esp32 ***@***.***>抄送： trberc ***@***.***>, Comment ***@***.***>主    题： Re: [78/xiaozhi-esp32] 请教下qwen实时模型是如何实现的 (Issue #605)aeolus21 left a comment (78/xiaozhi-esp32#605)

用火山的tts 基本在300ms 很快。llm用 阿里新出的qwen3基础上的 truber-latest模型也很快一般问题llm+tts 在500-700ms左右，现在主要是asr 我用的豆包，还可以吧总的下来1s左右发自我的手机-------- 原始邮件 --------发件人： aeolus21 @.>日期： 2025年7月11日周五 中午11:31收件人： 78/xiaozhi-esp32 @.>抄送： trberc @.>, Comment @.>主    题： Re: [78/xiaozhi-esp32] 请教下qwen实时模型是如何实现的 (Issue #605)aeolus21 left a comment (78/xiaozhi-esp32#605)
我这边也自己基于开源server搭了一套，tts也用的豆包的湾湾小何，延时也很大，楼主后来解决了吗

您说的火山tts是HuoshanDoubleStreamTTS 双流式吗