78/xiaozhi-esp32 #608: 如何与另一块ESP32S3进行通信

baikal2411, 2025-05-13:
如果选用ATOMS3-M12-CAM，想令其与另一块ESP32S3N16R8进行通信，如果考虑使用ESPNOW协议，应该对代码如何修改呢？谢谢！

baikal2411, 2025-05-14:
[image]

[image]
在application.cc当中编写如上的espnow的配置代码，结果与已有的wifi配置冲突，卡在这里：

[image]
就无法进行下去了

sunq945, 2025-05-18:
esp_err_t Set_Wifi_Channel(uint8_t _primaryChan)
{
    // 断开连接
    esp_err_t ret = esp_wifi_disconnect();
    if (ret != ESP_OK) {
        ESP_LOGE("wifi", "Failed to disconnect: %d", ret);
        return ret;
    }
    
    
    wifi_second_chan_t secondChan = WIFI_SECOND_CHAN_NONE;
    ESP_ERROR_CHECK(esp_wifi_set_channel(_primaryChan, secondChan));    
    
    // 重新连接
    ret = esp_wifi_connect();
    if (ret != ESP_OK) {
        ESP_LOGE("wifi", "Failed to reconnect: %d", ret);        
    }  
    return ret;
}

在wifi_board.cc中的void WifiBoard::StartNetwork()函数中，加入以上函数：

[image]

在wifi_board.h中加入CONFIG_ESPNOW_CHANNEL的定义，具体通道号自己定义（根据官方的文档我记得可选0~13）

#ifndef CONFIG_ESPNOW_CHANNEL
#define CONFIG_ESPNOW_CHANNEL 6
#endif