78/xiaozhi-esp32 #610: ml307联网，mqtt推送消息的时候数组过大导致失败

fenyoyo, 2025-05-13:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.6.2_bread-compact-ml307

### Operating System used.

Windows

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

通过mqtt, 发布消息的的时候， iot的descriptors可以上传成功

### What is the actual behavior?

通过mqtt,，当iot的descriptors数组过大时，就会出现MQTT: Failed to publish message

### Steps to reproduce.

1. 修改lamp.cc
不断的添加AddMethod，添加到10个左右，就会出现这个问题,其他的也是一样，一旦属性或这个方法到10个左右，都会导致发送失败。

```
#include "iot/thing.h"
#include "board.h"
#include "audio_codec.h"

#include <driver/gpio.h>
#include <esp_log.h>

#define TAG "Lamp"

namespace iot
{

    // 这里仅定义 Lamp 的属性和方法，不包含具体的实现
    class Lamp : public Thing
    {
    private:
#ifdef CONFIG_IDF_TARGET_ESP32
        gpio_num_t gpio_num_ = GPIO_NUM_12;
#else
        gpio_num_t gpio_num_ = GPIO_NUM_18;
#endif
        bool power_ = false;

        void InitializeGpio()
        {
            gpio_config_t config = {
                .pin_bit_mask = (1ULL << gpio_num_),
                .mode = GPIO_MODE_OUTPUT,
                .pull_up_en = GPIO_PULLUP_DISABLE,
                .pull_down_en = GPIO_PULLDOWN_DISABLE,
                .intr_type = GPIO_INTR_DISABLE,
            };
            ESP_ERROR_CHECK(gpio_config(&config));
            gpio_set_level(gpio_num_, 0);
        }

    public:
        Lamp() : Thing("Lamp", "一个测试用的灯"), power_(false)
        {
            InitializeGpio();

            // 定义设备的属性
            properties_.AddBooleanProperty("power", "灯是否打开", [this]() -> bool
                                           { return power_; });

            // 定义设备可以被远程执行的指令
            methods_.AddMethod("TurnOn", "打开灯", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = true;
            gpio_set_level(gpio_num_, 1); });

            methods_.AddMethod("TurnOff", "关闭灯2", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = false;
            gpio_set_level(gpio_num_, 0); });
            methods_.AddMethod("TurnOff2", "关闭灯3", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = false;
            gpio_set_level(gpio_num_, 0); });
            methods_.AddMethod("TurnOff3", "关闭灯4", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = false;
            gpio_set_level(gpio_num_, 0); });
            methods_.AddMethod("TurnOff4", "关闭灯5", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = false;
            gpio_set_level(gpio_num_, 0); });
            methods_.AddMethod("TurnOff5", "关闭灯6", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = false;
            gpio_set_level(gpio_num_, 0); });
            methods_.AddMethod("TurnOff6", "关闭灯7", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = false;
            gpio_set_level(gpio_num_, 0); });
            methods_.AddMethod("TurnOff7", "关闭灯7", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = false;
            gpio_set_level(gpio_num_, 0); });
            methods_.AddMethod("TurnOff8", "关闭灯7", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = false;
            gpio_set_level(gpio_num_, 0); });
            methods_.AddMethod("TurnOff9", "关闭灯7", ParameterList(), [this](const ParameterList &parameters)
                               {
            power_ = false;
            gpio_set_level(gpio_num_, 0); });
        }
    };

} // namespace iot

DECLARE_THING(Lamp);

```

### Debug Logs.

```plain
I (57472) MQTT: publish message: {"type":"hello","version": 3,"transport":"udp","audio_params":{"format":"opus", "sample_rate":16000, "channels":1, "frame_duration":60}}
E (60952) MQTT: Failed to publish message: {"session_id":"fe6f457e","type":"iot","update":true,"descriptors":[{"name":"Lamp","description":"一个测试用的灯","properties":{"power":{"description":"灯是否打开","type":"boolean"}},"methods":{"TurnOn":{"description":"打开灯","parameters":{}},"TurnOff":{"description":"关闭灯2","parameters":{}},"TurnOff2":{"description":"关闭灯3","parameters":{}},"TurnOff3":{"description":"关闭灯4","parameters":{}},"TurnOff4":{"description":"关闭灯5","parameters":{}},"TurnOff5":{"description":"关闭灯6","parameters":{}},"TurnOff6":{"description":"关闭灯7","parameters":{}},"TurnOff11":{"description":"关闭灯7","parameters":{}},"TurnOff16":{"description":"关闭灯7","parameters":{}},"TurnOff26":{"description":"关闭灯7","parameters":{}},"TurnOff36":{"description":"关闭灯7","parameters":{}},"TurnOff46":{"description":"关闭灯7","parameters":{}},"TurnOff426":{"description":"关闭灯7","parameters":{}},"TurnOff446":{"description":"关闭灯7","parameters":{}},"TurnOff416":{"description":"关闭灯7","parameters":{}}}}]}
E (64532) MQTT: Failed to publish message: {"session_id":"fe6f457e","type":"iot","update":true,"descriptors":[{"name":"客厅-米家落地扇","description":"米家落地扇","properties":{"fan:on":{"description":"电源","type":"boolean"},"fan:fan-level":{"description":"风速","type":"number"},"fan:horizontal-swing":{"description":"扫风","type":"boolean"},"fan:mode":{"description":"模式","type":"number"}},"methods":{"On":{"description":"电源","parameters":{"value":{"description":"","type":"boolean"}}},"FanLevel":{"description":"风速","parameters":{"value":{"description":"1-3档","type":"number"}}},"HorizontalSwing":{"description":"扫风","parameters":{"value":{"description":"","type":"boolean"}}},"Mode":{"description":"模式","parameters":{"value":{"description":"0=直吹风|1=睡眠风","type":"number"}}}}}]}
```

78, 2025-05-13:
There's a size limitation in ML307 MQTT packet. Around 2KB, IIRC.
Try to combine similar methods into one like `TurnOff(light_number)` and set how many lights in the method description.

fenyoyo, 2025-05-14:
{"session_id":"fe6f457e","type":"iot","update":true,"descriptors":[{"name":"客厅-米家落地扇","description":"米家落地扇","properties":{"fan:on":{"description":"电源","type":"boolean"},"fan:fan-level":{"description":"风速","type":"number"},"fan:horizontal-swing":{"description":"扫风","type":"boolean"},"fan:mode":{"description":"模式","type":"number"}},"methods":{"On":{"description":"电源","parameters":{"value":{"description":"","type":"boolean"}}},"FanLevel":{"description":"风速","parameters":{"value":{"description":"1-3档","type":"number"}}},"HorizontalSwing":{"description":"扫风","parameters":{"value":{"description":"","type":"boolean"}}},"Mode":{"description":"模式","parameters":{"value":{"description":"0=直吹风|1=睡眠风","type":"number"}}}}}]}  
这个数据中，上传的是设备电源状态，扫风模式，风速和模式，以及对应的控制方法。这个数据没有到达1KB,但也是失败的。

fenyoyo, 2025-05-14:
这里增加了一行调试信息，输出得到的结果
```
    if (!mqtt_->Publish(publish_topic_, text))
    {
        ESP_LOGE(TAG, "Failed to publish message size: %d", text.size());
        ESP_LOGE(TAG, "Failed to publish message: %s", text.c_str());
        SetError(Lang::Strings::SERVER_ERROR);
        return false;
    }
```

```
E (30412) MQTT: Failed to publish message size: 764
E (30412) MQTT: Failed to publish message: {"session_id":"afd1c28e","type":"iot","update":true,"descriptors":[{"name":"Lamp","description":"一个测试用的灯","properties":{"power":{"description":"灯是否打开","type":"boolean"}},"methods":{"TurnOn":{"description":"打开灯","parameters":{}},"TurnOff":{"description":"关闭灯2","parameters":{}},"TurnOff2":{"description":"关闭灯3","parameters":{}},"TurnOff3":{"description":"关闭灯4","parameters":{}},"TurnOff4":{"description":"关闭灯5","parameters":{}},"TurnOff5":{"description":"关闭灯6","parameters":{}},"TurnOff6":{"description":"关闭灯7","parameters":{}},"TurnOff7":{"description":"关闭灯7","parameters":{}},"TurnOff8":{"description":"关闭灯7","parameters":{}},"TurnOff9":{"description":"关闭灯7","parameters":{}}}}]}
```

78, 2025-05-14:
> 这里增加了一行调试信息，输出得到的结果
>
> ```
>     if (!mqtt_->Publish(publish_topic_, text))
>     {
>         ESP_LOGE(TAG, "Failed to publish message size: %d", text.size());
>         ESP_LOGE(TAG, "Failed to publish [...]

根据手册，ML302A/ML305A/ML307A/ML302S/ML307S长度最大为1024。由于当前组件用了HEX编码，所以实际为512。修改编码可能是最快的扩容方式。

fenyoyo, 2025-05-14:
请问  该如何调整呢

galoisgroupcn, 2025-05-15:
@fenyoyo try this. https://github.com/galoisgroupcn/xiaozhi-esp32/pull/4

78, 2025-05-20:
https://github.com/78/esp-ml307/commit/3ddaac1d29803e82e81f9f33a259650e04ba448b

esp-ml307组件已经提交了新的commit，支持包大小从750扩展到1450字节。

closed 2025-05-20 by 78