78/xiaozhi-esp32 #619: Location based on users internet connection.

RealDeco, 2025-05-15:
I wouldn't call it a bug, but i tried to override the location it gets from the internet provider (which in my country isn't always the same region), so asking "what is todays weather report?" it would give me one for my internet providers region, not my actual region.

thats all good if nothing else was specified. 

adding to the prompt "your location is Copenhagen, Denmark" doesn't change that, it still ignores the prompt and use internet provider location (Aarhus, Denmark). 

shouldn't the prompt override the location it reads from internet provider?

i can TELL it where i really am and it promises to remember that, but we all know thats not forever ;)

78, 2025-05-20:
Currently, under a Wi-Fi network, the weather is obtained based on the IP address location, while under a cellular network, it supports using the location determined by cell tower positioning to fetch weather information.

You can use a prompt like this: when asked about the weather, always query the weather information for Copenhagen, Denmark.

RealDeco, 2025-05-21:
thank you for the reply and yes a better prompt helped :)

closed 2025-05-21 by RealDeco