78/xiaozhi-esp32 #625: 小智 IOT 操作时的语音控制问题

jiangyanfeng, 2025-05-16:
### Is your feature request related to a problem?

能否让小智在语音识别后， IOT 调用对应函数的时候，播放完输出内容后，不再切换为 Listening 状态？
目前在做本地 TF 卡音乐的播放，如果直接在 IOT 响应函数里切换状态的话，会导致小智的语音播放不出来。

另外，可否设计小智在关闭服务器连接前不要再语音通知？因为本地 TF 卡音乐播放会被打断。

希望能提供接口实现上述功能。

78, 2025-05-16:
建议在你的IoT函数里直接CloseAudioChannel。

jiangyanfeng, 2025-05-17:
> 建议在你的IoT函数里直接CloseAudioChannel。

可以，试了下效果达到。麻烦在 Application.h 中添加对应接口：

```
    void CloseAudioInput() {
        protocol_->CloseAudioChannel();
    }
```

78, 2025-05-17:
也可以判断如果当前 device state 是 listening，则调用一下 Application::ToggleChatState

jiangyanfeng, 2025-05-17:
> 也可以判断如果当前 device state 是 listening，则调用一下 Application::ToggleChatState
对，谢谢提醒。
如果能增加状态改变回调就更好了。OnDeviceStateChanged()

bao17634, 2025-05-18:
你好麻烦请教一下你音乐播放器是不是用adf实现的

jiangyanfeng, 2025-05-18:
> 你好麻烦请教一下你音乐播放器是不是用adf实现的

是的。碰到的坑太多了，有些到现在还没填平。ADF 里面有些库不是开源的，只提供了静态库文件和头文件，debug 起来可麻烦了。

另外，由于对ESP32硬件不熟悉，内存分配也走了很多弯路。std::map 的表现就和 x86 上的 std::map 不一样，也不敢对框架改太多，怕影响后面版本的合入。

bao17634, 2025-05-18:
> > 你好麻烦请教一下你音乐播放器是不是用adf实现的
> 
> 是的。碰到的坑太多了，有些到现在还没填平。ADF 里面有些库不是开源的，只提供了静态库文件和头文件，debug 起来可麻烦了。
> 
> 另外，由于对ESP32硬件不熟悉，内存分配也走了很多弯路。std::map 的表现就和 x86 上的 std::map 不一样，也不敢对框架改太多，怕影响后面版本的合入。

我现在就是这种感觉，只能新建文件，不敢改太多，C++我的基础也很薄弱

bao17634, 2025-05-25:
我在播放音乐的时候复用的音频复用的是audio_code功能，我在启用audio_code的时候然后里面就有个现成禁用了，导致一直无法播放

[image]

jiangyanfeng, 2025-05-26:
> > 也可以判断如果当前 device state 是 listening，则调用一下 Application::ToggleChatState
> > 对，谢谢提醒。
> > 如果能增加状态改变回调就更好了。OnDeviceStateChanged()

目前是使用下面代码实现的，但是会在切换时咔哒一声。如果有状态改变回调函数使用就好了。 @78

78, 2025-05-26:
你可以先自己加个回调功能。

bao17634, 2025-05-26:
> 为了避免和小智AI的说话冲突

老哥能加个qq或者微信吗，真的想跟你请教下，我现在就是在执行完lot断开连接的，但还是播放不了声音