78/xiaozhi-esp32 #627: waveshare-esp32-s3-touch-lcd-1.85 升级 v1.6.2 后背光问题

konglingboy, 2025-05-16:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.6.2_esp32-s3-touch-lcd-1.85

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

waveshare-esp32-s3-touch-lcd-1.85 升级 v1.6.2 后，重启设备后屏幕应该亮起保持。

### What is the actual behavior?

但实际上，启动后后短时间内就进入背光关闭状态。

### Steps to reproduce.

1. Step 升级到 v1.6.2
2. Step 重启设备

### Debug Logs.

```plain
重启后，boards/esp32-s3-touch-lcd-1.85/esp32-s3-touch-lcd-1.85.cc 中 iot_button_register_cb(pwr_btn, BUTTON_LONG_PRESS_START, nullptr, [](void* button_handle, void* usr_data){} 被触发，导致背光被设置为零。
```

78, 2025-05-16:
看起来这是个bug，可以提交一个PR吗？
我先关闭此版本的OTA。

konglingboy, 2025-05-17:
github的协作我没用过，不是太懂，可能无法提交正确的PR。我先学习一下，后续问题如果有解决方法我会试着先提交PR，这个我先关闭了。

closed 2025-05-17 by konglingboy

78, 2025-05-17:
@Sail-211010 Check this out.

Sail-211010, 2025-05-23:
目前是没有复现问题，应该是你这边处理过了哈？