78/xiaozhi-esp32 #631: PDM无法唤醒唤醒

xixixihaha, 2025-05-16:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.6.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

PDM 模式的麦克风也应该可以唤醒小智

### What is the actual behavior?

PDM 模式的麦克风无法唤醒小智

### Steps to reproduce.

1. 烧写最新版本固件到使用PDM麦克风的硬件
2. 正常连接和激活
3. 发现无法语音唤醒
4. 按boot0键 进入唤醒状态，发现可以正常和小智交互，LED也可以正常反映MIC数据读写正常，说话红灯高亮 不说话红灯弱
...

### Debug Logs.

```plain

```

78, 2025-05-20:
有部分板子也是用PDM mic的，可以参考一下。

xixixihaha, 2025-05-21:
我目前的发现是刚启动的时候无法唤醒，多等一段时间就可以唤醒了，具体原因还在查