78/xiaozhi-esp32 #636: 用ADF实现了个流式播放音乐的功能，但是音质很差

bao17634, 2025-05-18:
### Is your feature request related to a problem?

用ADF实现了个流式播放音乐的功能，但是音质很差，后端服务将url传到硬件，然后硬件硬件采用esp-adf 通过url流式播放，播放的音质很差，请教下有办法解决吗

jiangyanfeng, 2025-05-18:
有多差？我觉得还行。
我也是用 ESP-ADF 杂交 ESP-IDF 的，感觉音质还行，但肯定别和立体声的播放器比了。
参见 https://www.bilibili.com/video/BV1JJEczPEVJ/?spm_id_from=333.1387.homepage.video_card.click

注意重采样成单声道、采样率和 codec 一样。

bao17634, 2025-05-18:
B站这个是你实现的吗？

jiangyanfeng, 2025-05-18:
对，上传视频时自动加上背景音乐了，导致原声不太清楚，实际上播放效果还行。
我这个采样是24000，单声道。
如果硬件能支持48000, 16bit,  双声道就好了。

bao17634, 2025-05-18:
可以向你请教下吗，我在b站留了言

jiangyanfeng, 2025-05-22:
> 可以向你请教下吗，我在b站留了言

已回复过了。

这个采样率改 48000 时，MP3播放倒是音质更好了，但小智语音回答就像快进的那种感觉，虽然也能听懂，但能否兼容一下 48000 采样率？我知道源是 16000 采样率的。目前codec设置 24000 没问题，但对MP3音质有一点影响。

78, 2025-08-25:
> > > 为啥我esp-adf播放音乐是慢放呢？
> > 
> > 
> > 采样率需要和 codec 一样。
> 
> 我现在想要改成全部44100采样。但是看着小智本身都是16000.不知道哪些地方需要改。还得需要研究一下源码才行啊

在 `boards/你的开放板目录名` 里的 `config.h` 定义了硬件输入输出的采样率