78/xiaozhi-esp32 #642: 电脑和小智连接的同一个wifi，ip的网段却不一样

bao17634, 2025-05-19:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v.1.2.1-bread-compact-wifi

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

电脑和小智的硬件连接的是同一个wifi，但是他们的网段却不一样，电脑的是192.168.3.14，小智的是192.168.5.112，导致小智无法连接到后台服务，我编译的是最新的代码。

[image]

### What is the actual behavior?

ip地址应该为同一个网段都是192.168.3.x

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

jiangyanfeng, 2025-05-19:
这个IP是DHCP分配的，应该检查下你的网络DHCP服务设置有什么特别的地方吧？

bao17634, 2025-05-20:
> 这个IP是DHCP分配的，应该检查下你的网络DHCP服务设置有什么特别的地方吧？

不是 是链接了我另外一个屋子的wifi，但是那个屋子的WiFi名字和我给小智链接的不一样，只是密码一样罢了

closed 2025-05-20 by bao17634