78/xiaozhi-esp32 #664: 无法连接到服务器，请稍后再试

sjysc, 2025-05-23:
esp32 wifi版WiFi已连接，但显示”无法连接到服务器，请稍后再试“，且无法通过按钮重置WiFi。
监视器部分输出如下：
I (3459) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3479) wifi:<ba-add>idx:0 (ifx:0, 50:fa:84:81:e6:9a), tid:0, ssn:0, winSize:64
I (3549) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4479) wifi: Got IP: 192.168.2.160
I (4479) esp_netif_handlers: sta ip: 192.168.2.160, mask: 255.255.255.0, gw: 192.168.2.1
I (4539) MQTT: Connecting to endpoint mqtt.xiaozhi.me
I (4619) wifi:<ba-add>idx:1 (ifx:0, 50:fa:84:81:e6:9a), tid:6, ssn:2, winSize:64
E (14549) MQTT: Failed to connect to endpoint
I (14549) Application: STATE: idle
I (14569) AudioCodec: Set output enable to false
W (14589) Application: Alert 错误: 无法连接服务，请稍后再试 [sad]
I (14599) AudioCodec: Set output enable to true
I (14599) Application: Resampling audio from 16000 to 24000
I (14599) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000
W (14619) esp-tls: Failed to open new connection in specified timeout
E (14619) transport_base: Failed to open a new connection
E (14629) mqtt_client: Error transport connect
I (14629) esp_mqtt: MQTT error occurred: ESP_ERR_ESP_TLS_CONNECTION_TIMEOUT
I (14649) main_task: Returned from app_main()
I (14649) Ota: Current version: 1.5.6
I (14649) EspHttp: Opening HTTP connection to https://api.tenclass.net/xiaozhi/ota/
I (14839) esp-x509-crt-bundle: Certificate validated
I (15699) Ota: Current is the latest version
I (15699) Ota: Running from factory partition, skipping
I (24649) Application: Free internal: 110572 minimal internal: 67560
I (27769) AudioCodec: Set output enable to false
I (34649) Application: Free internal: 108132 minimal internal: 67560
E (39659) esp-tls: [sock=54] select() timeout
E (39659) esp-tls: Failed to open new connection
E (39659) transport_base: Failed to open a new connection
E (39659) mqtt_client: Error transport connect
I (39669) esp_mqtt: MQTT error occurred: ESP_ERR_ESP_TLS_CONNECTION_TIMEOUT
I (44649) Application: Free internal: 110544 minimal internal: 67560
I (48469) wifi:state: run -> init (0x6c0)
I (48469) wifi:pm stop, total sleep time: 39463498 us / 45009520 us

78, 2025-05-23:
换个网络试试

MichaelDu9226, 2025-07-22:
E (23834) esp-tls: [sock=54] delayed connect error: Software caused connection abort
E (23834) esp-tls: Failed to open new connection
E (23834) transport_base: Failed to open a new connection
E (23834) HTTP_CLIENT: Connection failed, sock < 0
E (23844) EspHttp: Failed to perform HTTP request: ESP_ERR_HTTP_CONNECT
E (23844) Ota: Failed to open HTTP connection
E (52364) esp-tls: [sock=54] delayed connect error: Software caused connection abort
E (52364) esp-tls: Failed to open new connection
E (52364) transport_base: Failed to open a new connection
E (52364) HTTP_CLIENT: Connection failed, sock < 0
E (52374) EspHttp: Failed to perform HTTP request: ESP_ERR_HTTP_CONNECT
E (52374) Ota: Failed to open HTTP connection
E (90894) esp-tls: [sock=54] delayed connect error: Software caused connection abort
E (90894) esp-tls: Failed to open new connection
E (90894) transport_base: Failed to open a new connection
@78 看起来这个esp-tls错误挺常见的，感觉像是如果一个设备经常重启重连，就经常出现这个错误，然后过段时间又会恢复正常。
我已经有多个设备在过去一个月的开发测试过程中，出现类似问题了，一般过段时间就正常了

78, 2025-07-22:
有把 log level 改为 DEBUG 吗？这里还是看不出具体原因。