78/xiaozhi-esp32 #673: 希望添加太极优派开发版支持

crecywang, 2025-05-25:
### Is your feature request related to a problem?

太极小派出了一个amoled的新版，太极优派，屏幕和ESP32-S3-Touch-AMOLED-1.75一样，希望大佬纳入官方支持，商家自己弄的小智固件版本很老，而且有bug，也不开放配置文件，求大佬支持，感谢！
附原理图：
[taiji-原理图.zip](https://github.com/user-attachments/files/20430472/taiji-.zip)

crecywang, 2025-05-26:
@yihua-wang 求大佬适配，感谢。

crecywang, 2025-06-22:
> 我也刚买了这个。目前只提供了原理图，旧版本的xiaozhi固件能用不？他这个是3.5mm的耳机孔，xiaozhi目前支持的板子里面有哪款是一样的吗

旧版的也是耳机孔，但是旧版的不能用。
新版的用的amoled，和waveshare-s3-touch-amoled-1.75板子一样。

这种开发版，商家还要对适配文件保密，也是无语噢。
开放出来，不是还能促进销量么。。

crecywang, 2025-07-05:
> [#888](https://github.com/78/xiaozhi-esp32/pull/888) [@78](https://github.com/78) 有空请帮忙review

感谢大佬！！！

yihua-wang, 2025-07-07:
@crecywang  触摸这个我也不太清楚上层的业务 我只是单纯的适配了lvgl的触摸驱动
具体有啥功能不太清楚 是否可以对比一下其他开发板

crecywang, 2025-07-07:
> [@crecywang](https://github.com/crecywang) 触摸这个我也不太清楚上层的业务 我只是单纯的适配了lvgl的触摸驱动 具体有啥功能不太清楚 是否可以对比一下其他开发板

本来点一下屏幕，可以触发小智，显示聆听中。现在点屏幕好像没有任何反应。
这个没有实体按钮的，我看大佬适配的太极小派的配置文件里面没有button相关配置的。
如果不是触摸失效的话，是不是这个button的命令，导致屏幕触发小智失效了呀。

crecywang, 2025-07-07:
> [@crecywang](https://github.com/crecywang) 更新了 再试试 [#888](https://github.com/78/xiaozhi-esp32/pull/888)
更新后的代码，编译报错：
/Users/wp/xiaozhi-esp32/main/boards/viewe-smartring-s3-amoled-1.75/esp32-s3-viewe-amoled-1.75.cc: In constructor 'VieweEsp32s3TouchAMOLED1inch75::VieweEsp32s3TouchAMOLED1inch75()':
/Users/wp/xiaozhi-esp32/main/boards/viewe-smartring-s3-amoled-1.75/esp32-s3-viewe-amoled-1.75.cc:322:53: error: 'BOOT_BUTTON_GPIO' was not declared in this scope
  322 |     VieweEsp32s3TouchAMOLED1inch75() : boot_button_(BOOT_BUTTON_GPIO) {
      |                                                     ^~~~~~~~~~~~~~~~
[2030/2037] Building CXX object esp-id.../audio_processing/afe_wake_word.cc.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/wp/xiaozhi-esp32/build/log/idf_py_stderr_output_6032 and /Users/wp/xiaozhi-esp32/build/log/idf_py_stdout_output_6032

yihua-wang, 2025-07-07:
@crecywang 应该是你的代码同步问题 可以重新pick一遍

crecywang, 2025-07-07:
> [@crecywang](https://github.com/crecywang) 应该是你的代码同步问题 可以重新pick一遍

刚重新试了一下，编译成功，输入测试成功，感谢大佬🙏