78/xiaozhi-esp32 #715: 固件1.6.6启动报错 E (199) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

tang9999, 2025-05-29:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

1.6.6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Power Supply used.

USB

### What is the expected behavior?

idf.py build + flash 刷新固件1.6.6 应该正常启动  
硬件型号1：Movecall Moji
硬件型号2：无名科技 1.54（ML307）

### What is the actual behavior?

下载1.6.6 用 idf.py build 正常，flash 刷新固件1.6.6后，硬件型号1：Movecall Moji 硬件型号2：无名科技 1.54（ML307）都在启动过程报错：
E (199) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

abort() was called at PC 0x420e102b on core 0
.--- 0x420e102b: check_i2c_driver_conflict at E:/Espressif/frameworks/esp-idf-v5.3.2/components/driver/i2c/i2c.c:1719

Backtrace: 0x40379ff2:0x3fceb1c0 0x40385589:0x3fceb1e0 0x4038de65:0x3fceb200 0x420e102b:0x3fceb270 0x42004467:0x3fceb290 0x420044de:0x3fceb2b0 0x40379b8e:0x3fceb2d0 0x403cc9c5:0x3fceb340 0x403cccbb:0x3fceb380 0x403c8959:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
...--- 0x40379ff2: panic_abort at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/panic.c:463
0x40385589: esp_system_abort at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/port/esp_system_chip.c:92
0x4038de65: abort at E:/Espressif/frameworks/esp-idf-v5.3.2/components/newlib/abort.c:38
0x420e102b: check_i2c_driver_conflict at E:/Espressif/frameworks/esp-idf-v5.3.2/components/driver/i2c/i2c.c:1719
0x42004467: do_global_ctors at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/startup.c:101
0x420044de: start_cpu0_default at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/startup.c:201
0x40379b8e: call_start_cpu0 at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/port/cpu_start.c:842 (discriminator 1)
0x40045c01: ets_run_flash_bootloader in ROM
0x40043ab6: main in ROM
0x40034c45: .stack_ok in ROM

ELF file SHA256: b5bcbcc6e

Rebooting...

### Steps to reproduce.

idf.py set-target esp32s3 
idf.py menuconfig
idf.py build
idf.py -p COM3 flash monitor

硬件型号1：Movecall Moji   
硬件型号2：无名科技 1.54（ML307）
都在166刷完的启动过程报错

### Debug Logs.

```plain
E (199) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

abort() was called at PC 0x420e102b on core 0
.--- 0x420e102b: check_i2c_driver_conflict at E:/Espressif/frameworks/esp-idf-v5.3.2/components/driver/i2c/i2c.c:1719

Backtrace: 0x40379ff2:0x3fceb1c0 0x40385589:0x3fceb1e0 0x4038de65:0x3fceb200 0x420e102b:0x3fceb270 0x42004467:0x3fceb290 0x420044de:0x3fceb2b0 0x40379b8e:0x3fceb2d0 0x403cc9c5:0x3fceb340 0x403cccbb:0x3fceb380 0x403c8959:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
...--- 0x40379ff2: panic_abort at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/panic.c:463
0x40385589: esp_system_abort at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/port/esp_system_chip.c:92
0x4038de65: abort at E:/Espressif/frameworks/esp-idf-v5.3.2/components/newlib/abort.c:38
0x420e102b: check_i2c_driver_conflict at E:/Espressif/frameworks/esp-idf-v5.3.2/components/driver/i2c/i2c.c:1719
0x42004467: do_global_ctors at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/startup.c:101
0x420044de: start_cpu0_default at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/startup.c:201
0x40379b8e: call_start_cpu0 at E:/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/port/cpu_start.c:842 (discriminator 1)
0x40045c01: ets_run_flash_bootloader in ROM
0x40043ab6: main in ROM
0x40034c45: .stack_ok in ROM

ELF file SHA256: b5bcbcc6e

Rebooting...
```

whble, 2025-05-29:
用IDF 5.4以上版本就

tang9999, 2025-05-29:
> 可能是你idf版本的问题，到idf.py menuconfig -> Component config ->Audio Codec Device Configuration里看一下Enable向后兼容I2C驱动程序（强制使用v5.3以上的旧i2c_driver）这个选项。

已经勾选这一选项，重新BUILD FLASH, 再启动故障没有消失！

78, 2025-05-30:
如果你没有使用到摄像头，可以删除 esp-camera 组件，重新build。

closed 2025-05-30 by tang9999