78/xiaozhi-esp32 #718: 立创实战派ESP32-S3升级到1.6.6后黑屏无响应

nilescc-1, 2025-05-29:
立创实战派ESP32-S3开发板，下载官方1.6.6版源码，vscode编译烧录后，无法启动，黑屏无响应。
下载实战派1.6.6固件，用Flash软件烧录，可以正常开启。

RealDeco, 2025-05-30:
i got the same problem with latest pull, after flashing it just freezes, tried with the compiled bin v1.6.6_bread-compact-wifi and it works fine (but without display, i need to compile for that to work), but when compiling it myself set to same board (default breadboard wifi) it freezes.

in the monitor it keeps looping this:

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379de8
--- 0x40379de8: esp_restart_noos at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/port/soc/esp32s3/system_internal.c:158

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x568
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc14
load:0x403cb700,len:0x2d3c
entry 0x403c8904
I (35) octal_psram: vendor id    : 0x0d (AP)
I (35) octal_psram: dev id       : 0x02 (generation 3)
I (36) octal_psram: density      : 0x03 (64 Mbit)
I (40) octal_psram: good-die     : 0x01 (Pass)
I (45) octal_psram: Latency      : 0x01 (Fixed)
I (51) octal_psram: VCC          : 0x01 (3V)
I (55) octal_psram: SRF          : 0x01 (Fast Refresh)
I (61) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (67) octal_psram: BurstLen     : 0x01 (32 Byte)
I (72) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (78) octal_psram: DriveStrength: 0x00 (1/1)
I (84) MSPI Timing: PSRAM timing tuning index: 5
I (89) esp_psram: Found 8MB PSRAM device
I (93) esp_psram: Speed: 80MHz
I (97) cpu_start: Multicore app
I (113) cpu_start: Pro cpu start user code
I (113) cpu_start: cpu freq: 240000000 Hz
I (113) app_init: Application information:
I (116) app_init: Project name:     xiaozhi
I (121) app_init: App version:      1.6.6
I (125) app_init: Compile time:     May 30 2025 12:49:50
I (131) app_init: ELF file SHA256:  c24f1f18a...
I (137) app_init: ESP-IDF:          v5.3.1-dirty
I (142) efuse_init: Min chip rev:     v0.0
I (147) efuse_init: Max chip rev:     v0.99
I (152) efuse_init: Chip rev:         v0.2
I (156) heap_init: Initializing. RAM available for dynamic allocation:
I (164) heap_init: At 3FCAD850 len 0003BEC0 (239 KiB): RAM
I (170) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (176) heap_init: At 600FE100 len 00001ED8 (7 KiB): RTCRAM
I (182) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (190) spi_flash: detected chip: generic
I (194) spi_flash: flash io: qio
E (199) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

abort() was called at PC 0x420d967b on core 0
--- 0x420d967b: check_i2c_driver_conflict at C:/Espressif/frameworks/esp-idf-v5.3.1/components/driver/i2c/i2c.c:1716

Backtrace: 0x40379eae:0x3fceb1c0 0x40384135:0x3fceb1e0 0x4038be35:0x3fceb200 0x420d967b:0x3fceb270 0x42004447:0x3fceb290 0x420044be:0x3fceb2b0 0x40379a4a:0x3fceb2d0 0x403cc941:0x3fceb340 0x403ccc37:0x3fceb380 0x403c8959:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
.--- 0x40379eae: panic_abort at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/panic.c:463
--- 0x40384135: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/port/esp_system_chip.c:92
--- 0x4038be35: abort at C:/Espressif/frameworks/esp-idf-v5.3.1/components/newlib/abort.c:38
--- 0x420d967b: check_i2c_driver_conflict at C:/Espressif/frameworks/esp-idf-v5.3.1/components/driver/i2c/i2c.c:1716
--- 0x42004447: do_global_ctors at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/startup.c:101
--- 0x420044be: start_cpu0_default at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/startup.c:201
--- 0x40379a4a: call_start_cpu0 at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/port/cpu_start.c:829 (discriminator 1)
--- 0x40045c01: ets_run_flash_bootloader in ROM
--- 0x40043ab6: main in ROM
--- 0x40034c45: .stack_ok in ROM

ELF file SHA256: c24f1f18a
.
Rebooting...

RealDeco, 2025-05-30:
thank you!, actually thought i was on latest esp-idf, now i just feel silly :)

nilescc-1, 2025-05-31:
感谢，已解决需要 ESP-IDF 版本升级到 5.4.0 以上

huangzhen123, 2025-07-28:
我也是这个板子，用这个项目用 ESP-IDF  5.4 刷完黑屏了