78/xiaozhi-esp32 #727: I2C驱动冲突

Boca24, 2025-05-30:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

commit 522

### Operating System used.

macOS

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

I (156) heap_init: Initializing. RAM available for dynamic allocation:
I (163) heap_init: At 3FCAD5C8 len 0003C148 (240 KiB): RAM
I (170) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (176) heap_init: At 600FE000 len 00001FD8 (7 KiB): RTCRAM
I (182) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (190) spi_flash: detected chip: boya
I (194) spi_flash: flash io: qio
**E (198) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver**

abort() was called at PC 0x420ac7b7 on core 0
--- 0x420ac7b7: check_i2c_driver_conflict at /Users/mini1u1/esp/v5.3.3/esp-idf/components/driver/i2c/i2c.c:1723

Backtrace: 0x40379eee:0x3fceb1c0 0x40383f41:0x3fceb1e0 0x4038bd6d:0x3fceb200 0x420ac7b7:0x3fceb270 0x420042eb:0x3fceb290 0x42004362:0x3fceb.
--- 0x40379eee: panic_abort at /Users/mini1u1/esp/v5.3.3/esp-idf/components/esp_system/panic.c:463
--- 0x40383f41: esp_system_abort at /Users/mini1u1/esp/v5.3.3/esp-idf/components/esp_system/port/esp_system_chip.c:87
--- 0x4038bd6d: abort at /Users/mini1u1/esp/v5.3.3/esp-idf/components/newlib/abort.c:38
--- 0x420ac7b7: check_i2c_driver_conflict at /Users/mini1u1/esp/v5.3.3/esp-idf/components/driver/i2c/i2c.c:1723
--- 0x420042eb: do_global_ctors at /Users/mini1u1/esp/v5.3.3/esp-idf/components/esp_system/startup.c:101
--- 0x42004362: start_cpu0_default at /Users/mini1u1/esp/v5.3.3/esp-idf/components/esp_system/startup.c:201

### What is the actual behavior?

系统停止启动

### Steps to reproduce.

每次上电复现

### Debug Logs.

```plain

```

78, 2025-05-31:
参考 https://github.com/78/xiaozhi-esp32/issues/718#issuecomment-2922459479

Boca24, 2025-05-31:
@参考 #718 (comment)

Two possible ways to solve E (199) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver problem.

Remove esp32-camera from idf_component.yml
Upgrade to ESP-IDF 5.4.1 or later versions and do a full clean rebuild (set-target or just remove the build folder)