78/xiaozhi-esp32 #746: 摄像头拍照识别报错

whosmyqueen, 2025-06-04:
@78 虾哥， 这个是怎么回事啊
[image]

78, 2025-06-04:
堆栈溢出了，这是最新的代码出现的吗？ 你是哪一款开发板。

whosmyqueen, 2025-06-04:
@78 这款芯片， OV2640摄像头

[image]

whosmyqueen, 2025-06-04:
`I (57345) Application: >> 看看前面是什么。
I (57355) Application: STATE: speaking
I (57825) Application: << 哎呀
I (58255) Esp32Camera: camera_fb_t: width -> 480, height -> 320, size -> 307200
W (58255) Esp32Camera: Skip preview because of unsupported frame size
I (58265) EspHttp: Opening HTTP connection to https://xiaozhi.whosmyqueen.top/mcp/vision/explain
I (61745) Application: << 前面是什么啊？
I (62305) Application: << 你这个问题好机车啊
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4038d7c2  PS      : 0x00060733  A0      : 0x8038cc48  A1      : 0x3fcbaae0
--- 0x4038d7c2: remove_free_block at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/tlsf/tlsf_control_functions.h:374
---  (inlined by) block_locate_free at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/tlsf/tlsf_control_functions.h:619
---  (inlined by) tlsf_memalign_offs at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/tlsf/tlsf.c:572

A2      : 0x3fcbb660  A3      : 0x00000008  A4      : 0x00000018  A5      : 0x00008000
A6      : 0x3fcbac84  A7      : 0xfffffffc  A8      : 0x00000004  A9      : 0x00000000
A10     : 0x00000404  A11     : 0x00000000  A12     : 0x3fcb9dac  A13     : 0x00000010
A14     : 0x00000001  A15     : 0x0000000f  SAR     : 0x00000014  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000410  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff
--- Error: ClearCommError failed (PermissionError(13, '设备不识别此命令。', None, 22))
--- Waiting for the device to reconnect--- 0x40056f5c: memcpy in ROM
--- 0x40056f72: memcpy in ROM

Backtrace: 0x4038d7bf:0x3fcbaae0 0x4038cc45:0x3fcbab00 0x4038cc92:0x3fcbab20 0x4037acbd:0x3fcbab40 0x4037adc1:0x3fcbab60 0x4037ab54:0x3fcbab90 0x4200542b:0x3fcbabb0 0x420f97fe:0x3fcbabd0 0x420f3d4a:0x3fcbac30 0x420f5a3d:0x3fcbac60 0x420f5a95:0x3fcbacc0 0x420f5b12:0x3fcbad00 0x420f5b6f:0x3fcbad40 0x420ea401:0x3fcbad90 0x420eac3a:0x3fcbadc0 0x420e1e5d:0x3fcbae00 0x420e219b:0x3fcbaed0 0x420e293e:0x3fcbaef0 0x420e29bc:0x3fcbaf30 0x420e2ace:0x3fcbaf50 0x420e5128:0x3fcbaf70 0x420dae2a:0x3fcbaf90 0x4217ec89:0x3fcbafb0 0x420245ca:0x3fcbafd0 0x4202421d:0x3fcbaff0 0x420244bd:0x3fcbb040 0x42142a01:0x3fcbb060 0x42009e1c:0x3fcbb090 0x403871d1:0x3fcbb0b0

--- 0x4038d7bf: remove_free_block at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/tlsf/tlsf_control_functions.h:373 (discriminator 1)
---  (inlined by) block_locate_free at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/tlsf/tlsf_control_functions.h:619 (discriminator 1)
---  (inlined by) tlsf_memalign_offs at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/tlsf/tlsf.c:572 (discriminator 1)
--- 0x4038cc45: multi_heap_aligned_alloc_impl_offs at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/multi_heap.c:285
--- 0x4038cc92: multi_heap_aligned_alloc_offs at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/multi_heap.c:32
--- 0x4037acbd: aligned_or_unaligned_alloc at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/heap_caps_base.c:85
--- 0x4037adc1: heap_caps_aligned_alloc_base at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/heap_caps_base.c:158
--- 0x4037ab54: heap_caps_aligned_alloc at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/heap_caps.c:535
--- 0x4200542b: heap_caps_aligned_calloc at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/heap/heap_caps.c:556
--- 0x420f97fe: esp_aes_process_dma at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/port/aes/dma/esp_aes_dma_core.c:1019
--- 0x420f3d4a: esp_aes_crypt_ctr at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:533
--- 0x420f5a3d: esp_aes_gcm_finish at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:587
--- 0x420f5a95: esp_aes_gcm_crypt_and_tag_partial_hw at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:645
--- 0x420f5b12: esp_aes_gcm_crypt_and_tag at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:748
--- 0x420f5b6f: esp_aes_gcm_auth_decrypt at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:774
--- 0x420ea401: mbedtls_cipher_aead_decrypt at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/mbedtls/library/cipher.c:1528
--- 0x420eac3a: mbedtls_cipher_auth_decrypt_ext at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/mbedtls/library/cipher.c:1674
--- 0x420e1e5d: mbedtls_ssl_decrypt_buf at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:1639
--- 0x420e219b: ssl_prepare_record_content at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:4093
--- 0x420e293e: ssl_get_next_record at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:5068
--- 0x420e29bc: mbedtls_ssl_read_record at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:4323
--- 0x420e2ace: mbedtls_ssl_read at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:5983
--- 0x420e5128: __wrap_mbedtls_ssl_read at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/mbedtls/port/dynamic/esp_ssl_tls.c:348
[... cut ...]
---  (inlined by) _M_run at C:/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_thread.h:253
--- 0x42142a01: execute_native_thread_routine at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/src/c++11/thread.cc:104 (discriminator 1)
--- 0x42009e1c: pthread_task_func at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/pthread/pthread.c:241
--- 0x403871d1: vPortTaskWrapper at C:/Users/zhengzhihui2.vendor/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139`

whosmyqueen, 2025-06-04:
@78  改成HVGA就会报这个错

[image]

78, 2025-06-04:
可能是因为你使用的是https连接，堆栈是不够用的。
试试在修改 `mcp_server.cc` 里的 4096 为 8192。

whosmyqueen, 2025-06-04:
@78 确实是这个原因，现在可以了

> 可能是因为你使用的是https连接，堆栈是不够用的。 试试在修改 `mcp_server.cc` 里的 4096 为 8192。

closed 2025-06-04 by whosmyqueen

78, 2025-06-04:
用6144好像也可以，我已经提交了一个commit，增加stack size