78/xiaozhi-esp32 #751: ShowActivationCode 在espsparkport 上不工作

poppinlr, 2025-06-04:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

git hash f5d6cd2d36576b31ccd95e3633619b7c47901590

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

屏幕应该显示
xiaozhi.me
837926 
[happy]
等信息

### What is the actual behavior?

[image]
设备已经进入激活流程,

[image]
但是屏幕没有反应

### Steps to reproduce.

运行最新代码遇到的这个显示问题

### Debug Logs.

```plain

```

78, 2025-06-04:
> PS:78/esp-ml307: ~2.1.5 找不到, 最多到78/esp-ml307: ~2.1.4

CDN还没更新过来，可以去 https://github.com/78/esp-ml307 这里拉取。

你试试关闭WeChat模式，应该能正常显示了？

78, 2025-06-04:
我测试了能正常显示激活码，检查你的屏幕是否有线路连接问题。

78, 2025-06-05:
应该去开源server的项目下提issue

poppinlr, 2025-06-05:
> 应该去开源server的项目下提issue

 设备没激活,server是正常的呀

poppinlr, 2025-06-06:
> 应该去开源server的项目下提issue

我把sdkconfig 删除,,重新生成屏幕正常显示了

closed 2025-06-06 by poppinlr