78/xiaozhi-esp32 #778: esp32唤醒词内存优化

wdmomoxx, 2025-06-06:
### Is your feature request related to a problem?

由于esp32的唤醒词模型还是老版本的wakenet9模型，测试内存占用好像比较大，如果不开启PSRAM，会造成内存访问出错，无限重启。大部分esp32模块都是不带PSRAM的，测试模型实际使用好像需要100k不到的内存空间？（必须连续？），理论上esp32的内存是够用的，AI调试没成功，虾哥能否优化一下这个内存问题吗？

mzxhzhp, 2025-06-09:
有个疑问，我看文档中说打包带时候有个 srmodel.bin 模型，需要烧写到指定的分区中。可是为啥我怎么都找不到这个文件在哪？使用 IDF 工具编译并烧录固件，设备确实有语音唤醒功能，那么模型文件是在什么时候从哪里加载的呢？

sun-xiangyu, 2025-06-10:
在 `build/srmodels` 文件夹下

sun-xiangyu, 2025-06-10:
> 由于esp32的唤醒词模型还是老版本的wakenet9模型，测试内存占用好像比较大，如果不开启PSRAM，会造成内存访问出错，无限重启。大部分esp32模块都是不带PSRAM的，测试模型实际使用好像需要100k不到的内存空间？（必须连续？），理论上esp32的内存是够用的，AI调试没成功，虾哥能否优化一下这个内存问题吗？

后续会支持 wakenet9s, 目前还在测试模型性能

mzxhzhp, 2025-06-10:
> 在 `build/srmodels` 文件夹下， [esp-sr/CMakeLists.txt ](https://github.com/espressif/esp-sr/blob/6964df0b554f99d62961dab15ca15865af1fe34d/CMakeLists.txt#L120-L143) 中定义了自动烧写逻辑

是不是可以理解为，当前项目编译固件的时候，会自动拉取 esp-sr 项目进来一起编译生成固件？可是，当前项目的分区表中并没有看到有啥特殊设置， model 其实位置都是一样的，如何能放下好几个模型。使用当前项目编译生成的固件可以正常工作，但却不需要修改分区表，莫非是 IDF 工具在哪里为那几个内置模型自动分配的分区？

mzxhzhp, 2025-06-10:
> > 在 `build/srmodels` 文件夹下
> 
> 通过 IDF 工具编译后生成的吗，意思是这个模型文件在 IDF 工具中提供的？可是我在 IDF 源码仓库中也没有找到。 问题是，AFE 不是内置了好几个模型嘛，有语音唤醒、端点检测、TTS 等。这些内置模型是看不见的，不知道往哪里写呢，如果用户要用自己的模型，写入 partition 分区岂不是会相互覆盖？

这个我好像知道了，模型文件的数据在 esp-sr 项目下面，编译时才打包，怪不得我没搜到。

78, 2025-06-10:
> ### Is your feature request related to a problem?
> 由于esp32的唤醒词模型还是老版本的wakenet9模型，测试内存占用好像比较大，如果不开启PSRAM，会造成内存访问出错，无限重启。大部分esp32模块都是不带PSRAM的，测试模型实际使用好像需要100k不到的内存空间？（必须连续？），理论上esp32的内存是够用的，AI调试没成功，虾哥能否优化一下这个内存问题吗？
> 

我没有esp32的板子，你可以参考 https://github.com/78/xiaozhi-esp32/commit/968ed1fae3707e71573143a497d0204d909ba7c3 这个对esp32c3的优化。

wdmomoxx, 2025-06-11:
@78 测试了比较困难，水平不行，哈哈哈，等待官方的wakenet9s模型吧

wdmomoxx, 2025-06-11:
@sun-xiangyu 等你的好消息