78/xiaozhi-esp32 #803: 像播报验证码一样播报Mac地址

Boca24, 2025-06-12:


xljeff, 2025-06-12:
已經實現使用很久, 但1.6.6之後MCP主導架構有比較大改變不一定適用現在版本, 可參考我的tool call

```
import json
from config.logger import setup_logging
from plugins_func.register import register_function, ToolType, ActionResponse, Action

TAG = __name__
logger = setup_logging()

GET_MAC_ADDRESS_FUNCTION_DESC = {
    "type": "function",
    "function": {
        "name": "get_mac_address",
        "description": (
            "獲取當前設備號碼。"
            "當用戶詢問“我的裝置編號是什麼？”、“裝置號碼”、“裝置編號”、“設備號碼”、“設備編號”或類似問題時使用此功能。"
        ),
        "parameters": {
            "type": "object",
            "properties": {
                "lang": {
                    "type": "string",
                    "description": "返回用戶使用的語言code，例如zh_TW/zh_HK/en_US/ja_JP等，默認zh_TW",
                }
            },
            "required": ["lang"],
        },
    },
}
```

```
@register_function("get_mac_address", GET_MAC_ADDRESS_FUNCTION_DESC, ToolType.SYSTEM_CTL)
def get_mac_address(conn, lang: str = "zh_TW"):
    """
    獲取並返回設備的MAC地址。
    MAC地址通常從連接時的 headers 中的 'device-id' 獲取。
    """
    try:
        device_id = conn.headers.get("device-id")
        client_ip = conn.client_ip # 只是記錄一下，非必要

        logger.bind(tag=TAG).info(f"請求獲取MAC地址。Device ID: {device_id}, Client IP: {client_ip}, Lang: {lang}")

        if device_id:
            # 準備給LLM的數據，讓LLM以自然的語言回應
            response_data_for_llm = (
                f"根據下列數據，用{lang}回答用戶關於本機MAC地址的查詢：\n\n"
                f"設備MAC地址: {device_id}\n\n"
                f"(請以自然、流暢的方式向用戶播報此MAC地址，例如：本機的MAC地址是 {device_id})"
            )
            return ActionResponse(Action.REQLLM, response_data_for_llm, None)
        else:
            logger.bind(tag=TAG).warning("無法從 headers 中獲取 device-id。")
            error_message_for_llm = (
                f"根據下列情況，用{lang}向用戶解釋無法獲取MAC地址：\n\n"
                f"錯誤: 無法獲取設備的MAC地址，可能是設備未提供此信息或系統配置問題。\n\n"
                f"(請以自然、友好的方式告知用戶此情況，例如：抱歉，我目前無法獲取到本機的MAC地址。)"
            )
            return ActionResponse(Action.REQLLM, error_message_for_llm, None)

    except Exception as e:
        logger.bind(tag=TAG).error(f"獲取MAC地址時發生錯誤: {e}")
        exception_message_for_llm = (
            f"根據下列情況，用{lang}向用戶解釋獲取MAC地址時發生錯誤：\n\n"
            f"錯誤: 系統在嘗試獲取MAC地址時遇到內部錯誤。\n\n"
            f"(請以自然、友好的方式告知用戶此情況，例如：抱歉，查詢MAC地址時發生了點小問題，請稍後再試。)"
        )
        return ActionResponse(Action.REQLLM, exception_message_for_llm, None)

```

Robotaxi-xo, 2025-06-12:
> 已經實現使用很久, 但1.6.6之後MCP主導架構有比較大改變不一定適用現在版本, 可參考我的tool call
>
> ```
> import json
> from config.logger import setup_logging
> from plugins_func.register import register_function, ToolType, ActionResponse, [...]

原來可以這樣搞, 我把問題複雜化了
感謝分享
全繁體, 也是台灣的朋友?

Boca24, 2025-06-12:
> 已經實現使用很久, 但1.6.6之後MCP主導架構有比較大改變不一定適用現在版本, 可參考我的tool call
>
> ```
> import json
> from config.logger import setup_logging
> from plugins_func.register import register_function, ToolType, ActionResponse, [...]

能否用IoT通过获取电池电量一样来实现。

78, 2025-06-12:
了解MCP：https://github.com/78/xiaozhi-esp32/blob/main/docs/mcp-usage.md