78/xiaozhi-esp32 #804: wvirgil123__esp_jpeg_simd 中的  audio_malloc.c 与 ESP-ADF 冲突

jiangyanfeng, 2025-06-12:
我使用了  ESP-ADF，编译时发现与 wvirgil123__esp_jpeg_simd 组件冲突。

D:/Users/JIANG/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/audio_sal/libaudio_sal.a(audio_mem.c.obj): in function `audio_free': 
D:/Users/JIANG/Documents/esp/esp-adf-master/components/audio_sal/audio_mem.c:94: multiple definition of `audio_free'; esp-idf/wvirgil123__esp_jpeg_simd/libwvirgil123__esp_jpeg_simd.a(audio_malloc.c.obj):D:/Users/JIANG/Documents/esp/xiaozhi-esp32/managed_components/wvirgil123__esp_jpeg_simd/src/audio_malloc.c:44: first defined here
D:/Users/JIANG/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/audio_sal/libaudio_sal.a(audio_mem.c.obj): in function `audio_calloc':

78, 2025-06-12:
> 我使用了 ESP-ADF，编译时发现与 wvirgil123__esp_jpeg_simd 组件冲突。 [...]

可以暂时注释掉那个组件。

Kevincoooool, 2025-06-12:
直接删掉这个组件，一般没人用的