78/xiaozhi-esp32 #806: 对设备说“休息”，设备无反应（不执行goodbye也不说话）

xujieuse, 2025-06-12:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.6.6_lichuang-dev

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

设备先回复，然后goodbye。（没有goodbye也没事，但不要什么也不做）

### What is the actual behavior?

设备不说话也不goodbye

### Steps to reproduce.

1. 对设备说“休息”
2. 多次尝试
...

### Debug Logs.

```plain
I (12511) Application: STATE: connecting
I (12591) MQTT: Session ID: fc1b58c6
I (12591) wifi:Set ps type: 0, coexist: 0

I (12591) Application: STATE: listening
I (15731) wifi:new:<11,2>, old:<11,0>, ap:<255,255>, sta:<11,2>, prof:1, snd_ch_cfg:0x0
I (15901) Application: >> 休息。
I (15901) Application: STATE: speaking
I (16031) Application: STATE: listening
I (16041) MQTT: Received goodbye message, session_id: fc1b58c6
I (16051) wifi:Set ps type: 1, coexist: 0

I (16061) Application: STATE: idle
I (20811) Application: STATE: connecting
I (20891) MQTT: Session ID: 039b8b2e
I (20891) wifi:Set ps type: 0, coexist: 0

I (20891) Application: STATE: listening
I (23941) Application: >> 休息。
I (23941) Application: STATE: speaking
I (24171) Application: STATE: listening
I (31691) Application: >> 休息。
I (31691) Application: STATE: speaking
I (31821) Application: STATE: listening
I (36801) Application: >> 休息。
I (36801) Application: STATE: speaking
I (36911) Application: STATE: listening
I (39801) wifi:new:<11,0>, old:<11,2>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
I (40721) wifi:new:<11,2>, old:<11,0>, ap:<255,255>, sta:<11,2>, prof:1, snd_ch_cfg:0x0
I (41131) Application: STATE: speaking
I (41141) Application: >> 休息。
I (41291) Application: STATE: listening
I (41501) MQTT: Received goodbye message, session_id: 039b8b2e
I (41501) wifi:Set ps type: 1, coexist: 0

I (41501) Application: STATE: idle
I (50971) SystemInfo: free sram: 32239 minimal sram: 28227
I (52161) I2S_IF: Pending out channel for in channel running
I (52161) AudioCodec: Set output enable to false
I (60971) SystemInfo: free sram: 32107 minimal sram: 28227
```

xujieuse, 2025-06-12:
[image]

78, 2025-06-12:
这个应该不算bug，系统提示词里确实没有休息这个指令，可以用再见、退下、待机等。

Boca24, 2025-06-12:
最近确实存在这个问题，无论换哪种方式说比如再见、退下，有时（5%概率）需要喊3次以上才收到goodbye

xujieuse, 2025-06-13:
> 最近确实存在这个问题，无论换哪种方式说比如再见、退下，有时（5%概率）需要喊3次以上才收到goodbye

你遇到的情况，应该和大模型function calling偷懒有关。

closed 2025-06-13 by xujieuse