78/xiaozhi-esp32 #828: error: 'class Ml307AtModem' has no member named 'SetSleepMode'

isamu2025, 2025-06-17:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v 1.7.5  ESP32-BOX

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

r/boards/common/ml307_board.cc.obj -c E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/main/boards/common/ml307_board.cc
E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/main/boards/common/ml307_board.cc: In member function 'void Ml307Board::WaitForNetworkReady()':
E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/main/boards/common/ml307_board.cc:70:12: error: 'class Ml307AtModem' has no member named 'SetSleepMode'
   70 |     modem_.SetSleepMode(true, 30);
      |            ^~~~~~~~~~~~
[2016/2025] Performing build step for 'bootloader'

Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/build/bootloader/bootloader.bin
[115/115] C:\Windows\system32\cmd.exe /C "cd /D E:\temp\xiaozhi\xiaozhi-esp32-1.7.5\build\bootloader\esp-idf\esptool_py && d:\tools\Espressif\vscode-tools\python_env\idf5.4_py3.11_env\Scripts\python.exe D:/tools/Espressif/esp/v5.4/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/build/bootloader/bootloader.bin"
Bootloader binary size 0x4040 bytes. 0x3fc0 bytes (50%) free.
ninja: build stopped: subcommand failed.

### What is the actual behavior?

r/boards/common/ml307_board.cc.obj -c E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/main/boards/common/ml307_board.cc
E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/main/boards/common/ml307_board.cc: In member function 'void Ml307Board::WaitForNetworkReady()':
E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/main/boards/common/ml307_board.cc:70:12: error: 'class Ml307AtModem' has no member named 'SetSleepMode'
   70 |     modem_.SetSleepMode(true, 30);
      |            ^~~~~~~~~~~~
[2016/2025] Performing build step for 'bootloader'

Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/build/bootloader/bootloader.bin
[115/115] C:\Windows\system32\cmd.exe /C "cd /D E:\temp\xiaozhi\xiaozhi-esp32-1.7.5\build\bootloader\esp-idf\esptool_py && d:\tools\Espressif\vscode-tools\python_env\idf5.4_py3.11_env\Scripts\python.exe D:/tools/Espressif/esp/v5.4/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 E:/temp/xiaozhi/xiaozhi-esp32-1.7.5/build/bootloader/bootloader.bin"
Bootloader binary size 0x4040 bytes. 0x3fc0 bytes (50%) free.
ninja: build stopped: subcommand failed.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

isamu2025, 2025-06-17:
xiaozhi-esp32-1.7.5>git pull
Updating f29c1a1..116234a
Fast-forward
 CMakeLists.txt         | 2 +-
 main/idf_component.yml | 2 +-
 2 files changed, 2 insertions(+), 2 deletions(-)

78, 2025-06-17:
Updating `78/esp-ml307: ~2.2.0` to `78/esp-ml307: ~2.2.1` solves this problem.

closed 2025-06-23 by 78