78/xiaozhi-esp32 #833: 立创s3烧录1.7.5 杂音严重

git-miao-i, 2025-06-19:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.7.5_lichuang-dev.zip

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

回复时声音清晰

### What is the actual behavior?

回复时 杂音很严重，说着说着就会chi chi la la 的杂音
立创 s3

### Steps to reproduce.

下载的压缩包固件 使用 flash 烧录

### Debug Logs.

```plain

```

78, 2025-06-19:
可以下载一段录音音频，上传来看看？

git-miao-i, 2025-06-19:
还是直接用手机录一个

git-miao-i, 2025-06-19:
就是那种回答着回答着就破音了

 

 

从QQ邮箱发来的超大附件

video_20250619_163611.mp4 (29.10M, 无限期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=79333166403116cdfc39532c44640a1c4f47520056070e551a52575506490a5504571c5f52505b1e5357530750060104560a545e627d38455e5754093d560801020307575b3b0905040500574c094807370e&code=731fbd83

video_20250619_165525.mp4 (36.34M, 无限期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=7c3361622ea48c9bf9390328173004494a47500653010e501f020250041d035302034c030953054b030b55560609525554000557312936105b57040d6e02065407035753086f0750070653571f5d4652320e&code=23ab106f

78, 2025-06-21:
奇怪，目前收到这个问题反馈的不多，你是在 GitHub releases下载的 1.7.4 版本吗？

Haugor, 2025-06-22:
可能和WIFI网络有一定关系，我之前是播放卡顿，换个热点就好了

78, 2025-06-23:
可以重新编译最新版本试试，或者等1.7.6版本，有可能是因为1.7.5开启了NSNET降噪，性能不足导致。

AndyLin2020, 2025-07-14:
换个功率大点的充电器试试

brucewwwww, 2025-09-30:
> 能是因为1.7.5开启了NSNET降噪，性能不足导致。

这个编译有选项吗，我在1.8.5的版本上有时候也会出这个问题。打印cpu线程信息"audio_input"高的时候是20多