78/xiaozhi-esp32 #840: 面包板ML307版本，通过MQTT协议给服务器发送json格式文本失败

harri8807, 2025-06-20:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.7.5

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

希望通过MQTT协议发给服务器一段文本，服务器帮我语音编码后回传播放

### What is the actual behavior?

从日志看服务器连接已成功且能正常唤醒小智语音助手并进行对话，但是调用protocol的SendText函数接口通过MQTT发送的json格式文本显示失败

### Steps to reproduce.

1. 4G模块连接小智服务
2. Wifi模块通过与本地某AP设备连接，通过websocket通信
3. 解析设备websocket回传的json内容并语音播放（将json内容发给小智服务进行文本语音编码）

### Debug Logs.

```plain
I (7921) MQTT: Connecting to endpoint mqtt.xiaozhi.me
I (7951) MQTT: Disconnected from endpoint
I (7951) Ml307Mqtt: MQTT connection state: 断开：用户主动断开
I (9201) Ml307Mqtt: MQTT connection state: 连接成功
I (9201) MQTT: Connected to endpoint
I (9211) MODEL_LOADER: The storage free size is 21952 KB
I (9211) MODEL_LOADER: The partition size is 960 KB
I (9211) MODEL_LOADER: Successfully load srmodels
I (9221) AFE: AFE Version: (1MIC_V250121)
I (9221) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (9231) AFE: AFE Pipeline: [input] -> |VAD(WebRTC)| -> [output]
I (9231) AfeAudioProcessor: Audio communication task started, feed size: 512 fetch size: 512
I (9241) AfeWakeWord: Model 0: wn9_nihaoxiaozhi_tts
I (9251) AFE_CONFIG: Set WakeNet Model: wn9_nihaoxiaozhi_tts
MC Quantized wakenet9: wakenet9l_tts1h8_你好小智_3_0.631_0.635, tigger:v4, mode:0, p:0, (Jun 16 2025 16:18:15)
I (9291) AFE: AFE Version: (1MIC_V250121)
I (9291) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (9301) AFE: AFE Pipeline: [input] ->  -> |VAD(WebRTC)| -> |WakeNet(wn9_nihaoxiaozhi_tts,)| -> [output]
I (9311) AfeWakeWord: Audio detection task started, feed size: 512 fetch size: 512
I (9321) Application: STATE: idle
I (9341) SystemInfo: free sram: 43531 minimal sram: 43435
I (9351) Application: Resampling audio from 16000 to 24000
I (9351) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000

...
I (118531) AudioCodec: Set output enable to true
E (118531) MQTT: Failed to publish message: { {
        "type": "hello",
        "version": 1,
        "features": {
            "mcp": true
        },
        "transport": "websocket",
        "audio_params": {
            "format": "opus",
            "sample_rate": 16000,
            "channels": 1,
            "frame_duration": 60
        }
       }
W (118561) Application: Alert 错误: 发送失败，请检查网络 [sad]
```

78, 2025-06-21:
为什么 mqtt 里会使用到 websocket？应该是 UDP 才对，这个是不是代码出了问题？

harri8807, 2025-06-24:
确实是我代码的问题，修复了以后通过MQTT可以发送并收到应答如下，我应该按照什么格式发送

[image]