78/xiaozhi-esp32 #845: ESP-Hi舵机小狗-1.聆听模式表情撕裂-2.舵机校准时零时不灵

DOCTOROLIVE, 2025-06-23:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

esp-hi 最近更新版本

### Operating System used.

Windows

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

CMD

### Power Supply used.

USB

### What is the expected behavior?

希望修复这两个BUG

### What is the actual behavior?

希望修复这两个BUG

### Steps to reproduce.

编译最新ESP-HI源码烧录，就能看到

### Debug Logs.

```plain
运行效果有问题
```

DOCTOROLIVE, 2025-07-01:
> ## 问题1
> （临时缓解方法）可先将 `idf_component.yml` 中的 `espressif2022/image_player` 版本降级至 1.1.0。稍后我们会调整该组件的适配。
> 
> [xiaozhi-esp32/main/idf_component.yml](https://github.com/78/xiaozhi-esp32/blob/d1c586239c9064c82b3162048cb8bee900446e0e/main/idf_component.yml#L31)
> 
> Line 31 in [d1c5862](/78/xiaozhi-esp32/commit/d1c586239c9064c82b3162048cb8bee900446e0e)
> 
>  espressif2022/image_player: ^1.1.0 
> 改为
> 
>   espressif2022/image_player: '==1.1.0'
> ## 问题2
> 可否补充更多细节？

针对问题二，就是网页校准的时候，当选中某个腿的时候，点击+ 或者 -的时候，对应舵机没有响应动作，感觉是舵机响应率低，跟不上 齿轮有卡卡声 但是不动作

DOCTOROLIVE, 2025-07-01:
> > 针对问题二，就是网页校准的时候，当选中某个腿的时候，点击+ 或者 -的时候，对应舵机没有响应动作，感觉是舵机响应率低，跟不上 齿轮有卡卡声 但是不动作
> 
> 如果齿轮有卡卡声说明程序已经将舵机动作指令发出去了，这个故障表现大概率是舵机问题，可以换个质量好点的舵机试试

换了金属舵机OK的，是不是舵机动作指令较快，塑料舵机的响应跟不上，查了资料，金属舵机的响应时间确实比塑料舵机少2倍以上@laride

wangzhihui0222-cpu, 2025-07-19:
> 发了一个新版的舵机控制组件，应该可以缓解舵机响应时间的问题。
> 
> 请尝试 `idf.py fullclean && rm dependencies.lock` 后重新编译；或将 `lijunru-hub/servo_dog_ctrl` 组件的版本指定为 `0.1.7` 再重新编译。（如果找不到这个版本，需要等组件仓库缓存更新）
> 
> 该版本还添加了组合技功能。
> 
> [@DOCTOROLIVE](https://github.com/DOCTOROLIVE)

请问一下，这个新版的舵机组件在哪，怎么直接用吗，可以说明一下吗？谢谢