78/xiaozhi-esp32 #846: 运行时，ESP32S3有时候会自动重启

goodtft, 2025-06-23:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

1.7.5

### Operating System used.

Windows

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

希望ESP32S3设备稳定运行，不会出现自动重启的情况。

### What is the actual behavior?

ESP32S3设备运行小智AI会自动重启。

### Steps to reproduce.

1. 上电运行，可以进行对话互动；
2. 然后对话结束，进入待命状态；
3. 过一段时间就会自动重启，自动重启间隔时间不固定；

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42091cf7  PS      : 0x00060430  A0      : 0x82073053  A1      : 0x3fcede60  
--- 0x42091cf7: dl_tie728_atrous_conv1d_accx_i16o16k16_check_ind_0 at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_atrous_conv1d_accx_i16o16k16.S:124

A2      : 0x3fceafb0  A3      : 0x3fceb170  A4      : 0x3c6d3b60  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000040  A9      : 0x00000003
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x3fceb0b0  A15     : 0x00000004  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x42091cf0  LEND    : 0x42091cfe  LCOUNT  : 0x00000000
--- 0x42091cf0: dl_tie728_atrous_conv1d_accx_i16o16k16_check_ind_0 at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_atrous_conv1d_accx_i16o16k16.S:122
--- 0x42091cfe: dl_tie728_atrous_conv1d_accx_i16o16k16_loop_i_end at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_atrous_conv1d_accx_i16o16k16.S:128

Backtrace: 0x42091cf4:0x3fcede60 0x42073050:0x3fcede70 0x4206f449:0x3fcede90 0x4206f4b8:0x3fcedeb0 0x4206f587:0x3fceded0 0x4218fc3e:0x3fcedef0 0x42056a9e:0x3fcedf20 0x42022753:0x3fcedf40 0x420227a5:0x3fcedf70 0x40384579:0x3fcedf90
--- 0x42091cf4: dl_tie728_atrous_conv1d_accx_i16o16k16_check_ind_0 at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_atrous_conv1d_accx_i16o16k16.S:123
--- 0x42073050: dl_nn_atrous_conv1d at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/nn/dl_nn_atrous_conv1d.c:134
--- 0x4206f449: wakenet9_model_run_step3 at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized.c:627
--- 0x4206f4b8: model_detect at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized.c:684
--- 0x4206f587: model_detect_mfcc at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized.c:715
--- 0x4218fc3e: afe_wn_run_single_mic at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:353
--- 0x42056a9e: afe_fetch_with_delay at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:615
--- 0x42022753: AfeWakeWord::AudioDetectionTask() at H:/project/7.0inch/ESP32/7.0inch_ESP32/xiaozhi-esp32_800x480/main/audio_processing/afe_wake_word.cc:122
--- 0x420227a5: AfeWakeWord::Initialize(AudioCodec*)::{lambda(void*)#1}::_FUN(void*) at H:/project/7.0inch/ESP32/7.0inch_ESP32/xiaozhi-esp32_800x480/main/audio_processing/afe_wake_word.cc:75
---  (inlined by) _FUN at H:/project/7.0inch/ESP32/7.0inch_ESP32/xiaozhi-esp32_800x480/main/audio_processing/afe_wake_word.cc:77
--- 0x40384579: vPortTaskWrapper at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: c44e96e23

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379ec1
--- 0x40379ec1: esp_restart_noos at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/port/soc/esp32s3/system_internal.c:160
```

78, 2025-06-23:
请提供一下有关设备信息和复现问题的方法

goodtft, 2025-06-24:
> 请提供一下有关设备信息和复现问题的方法

设备信息：
我在main/boards目录下添加了我自己的设备。我的设备是7寸显示屏，使用ESP32-S3驱动，分别率为800x480，接口为RGB565，不需要通过3线制SPI初始化。

复现方法：
连接服务器后，可以正常聊天。聊天结束后，进入待机状态，过一段时间就自动重启。这个重启间隔不是固定的，有时候等比较久才会自动重启，有时候很快就会自动重启。

ydcydc26, 2025-06-26:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379fcd
--- 0x40379fcd: panicHandler at /Users/yangdechen/esp/v5.4.1/esp-idf/components/esp_system/port/panic_handler.c:247

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x56c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc2c
load:0x403cb700,len:0x2e58
entry 0x403c890c 
我这有时候也会重启，版本1.6.6

78, 2025-06-26:
> ESP-ROM:esp32s3-20210327 Build:Mar 27 2021 rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT) Saved PC:0x40379fcd --- 0x40379fcd: panicHandler at [...]

请更新至新版本

linaiyue, 2025-07-02:
> > 可以尝试拉取最新的commit，看能否解决你的问题
> 
> 好的，我去试下，谢谢。

大佬，请问你拉取最新commit还有这个情况出现吗？

ydcydc26, 2025-07-02:
更新到1.76后没出现这个问题了

goodtft, 2025-07-17:
> 可以尝试拉取最新的commit，看能否解决你的问题
使用最新的版本（1.7.6）还是会自动重启，log：
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4208baec: 11ffc0ed 84051ef7 00861a08
--- 0x4208baec: dl_tie728_atrous_conv1d_accx_i16o16k16_loop_i_end at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_atrous_conv1d_accx_i16o16k16.S:128

Core  0 register dump:
PC      : 0x4208baf3  PS      : 0x00060230  A0      : 0x8206ce4f  A1      : 0x3fcead70
--- 0x4208baf3: dl_tie728_atrous_conv1d_accx_i16o16k16_loop_i_end at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_atrous_conv1d_accx_i16o16k16.S:132

A2      : 0x3fce8770  A3      : 0x3fce8930  A4      : 0x3c6b7460  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000080  A9      : 0x00000004
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000002  A13     : 0x00000000
A14     : 0x00000040  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4208badc  LEND    : 0x4208baea  LCOUNT  : 0x00000000
--- 0x4208badc: dl_tie728_atrous_conv1d_accx_i16o16k16_check_ind_0 at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_atrous_conv1d_accx_i16o16k16.S:122
--- 0x4208baea: dl_tie728_atrous_conv1d_accx_i16o16k16_loop_i_end at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_atrous_conv1d_accx_i16o16k16.S:128

Backtrace: 0x4208baf0:0x3fcead70 0x4206ce4c:0x3fcead80 0x42069245:0x3fceada0 0x420692b4:0x3fceadc0 0x42069383:0x3fceade0 0x42173e9a:0x3fceae00 0x4205089a:0x3fceae30 0x4201cad4:0x3fceae50 0x4201cb61:0x3fceae90
--- 0x4208baf0: dl_tie728_atrous_conv1d_accx_i16o16k16_loop_i_end at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_atrous_conv1d_accx_i16o16k16.S:130
--- 0x4206ce4c: dl_nn_atrous_conv1d at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/nn/dl_nn_atrous_conv1d.c:134
--- 0x42069245: wakenet9_model_run_step3 at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized.c:627
--- 0x420692b4: model_detect at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized.c:684
--- 0x42069383: model_detect_mfcc at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized.c:715
--- 0x42173e9a: afe_wn_run_single_mic at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:353
--- 0x4205089a: afe_fetch_with_delay at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:615
--- 0x4201cad4: AfeWakeWord::AudioDetectionTask() at H:/project/7.0inch/ESP32/7.0inch_ESP32/newest_xiaozhi_AI/xiaozhi-esp32-800x480/main/audio_processing/afe_wake_word.cc:122
--- 0x4201cb61: AfeWakeWord::Initialize(AudioCodec*)::{lambda(void*)#1}::_FUN(void*) at H:/project/7.0inch/ESP32/7.0inch_ESP32/newest_xiaozhi_AI/xiaozhi-esp32-800x480/main/audio_processing/afe_wake_word.cc:75
---  (inlined by) _FUN at H:/project/7.0inch/ESP32/7.0inch_ESP32/newest_xiaozhi_AI/xiaozhi-esp32-800x480/main/audio_processing/afe_wake_word.cc:77

ELF file SHA256: 124d77b6a

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379cfd
--- 0x40379cfd: esp_restart_noos at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/port/soc/esp32s3/system_internal.c:160

goodtft, 2025-07-24:
> 电压不足也会重启

一段时间来，发现了使用1.7.6版本还是会自动重启，具体情况如下：
长时间断电关机后再上电开机，前面大概半个消失内，会自动重启2、3次，然后长时间就不会自动重启了。断电关机时长大概是晚上8点到早上9点。短时间断电关机再上电开机不会有自动重启现象。串口输出的自动重启信息，有好几种：
第一种：

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42093073  PS      : 0x00070c30  A0      : 0x8206ce4f  A1      : 0x3fcead70
--- 0x42093073: dl_tie728_pointwise_conv1d_qacc_i8o8k8b8_preload_core_loop_i_end at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_pointwise_conv1d_qacc_i8o8k8b8_preload_core.S:128

A2      : 0x3fce8740  A3      : 0x3fce8900  A4      : 0x3c6d8630  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x000000c0  A9      : 0x00000006
A10     : 0x00000000  A11     : 0x00000000  A12     : 0xc208baf4  A13     : 0x00000040
A14     : 0x3fce8810  A15     : 0x00000004  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x42093028  LEND    : 0x4209306b  LCOUNT  : 0x00000000
--- 0x42093028: dl_tie728_pointwise_conv1d_qacc_i8o8k8b8_preload_core_loop_t at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_pointwise_conv1d_qacc_i8o8k8b8_preload_core.S:100
--- 0x4209306b: dl_tie728_pointwise_conv1d_qacc_i8o8k8b8_preload_core_loop_i_end at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_pointwise_conv1d_qacc_i8o8k8b8_preload_core.S:121

Backtrace: 0x42093070:0x3fcead70 0x4206ce4c:0x3fcead80 0x42069245:0x3fceada0 0x420692b4:0x3fceadc0 0x42069383:0x3fceade0 0x42173e9a:0x3fceae00 0x4205089a:0x3fceae30 0x4201cad4:0x3fceae50 0x4201cb61:0x3fceae90
--- 0x42093070: dl_tie728_pointwise_conv1d_qacc_i8o8k8b8_preload_core_loop_i_end at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_pointwise_conv1d_qacc_i8o8k8b8_preload_core.S:124
--- 0x4206ce4c: dl_nn_atrous_conv1d at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/nn/dl_nn_atrous_conv1d.c:134
--- 0x42069245: wakenet9_model_run_step3 at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized.c:627
--- 0x420692b4: model_detect at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized.c:684
--- 0x42069383: model_detect_mfcc at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized.c:715
--- 0x42173e9a: afe_wn_run_single_mic at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:353
--- 0x4205089a: afe_fetch_with_delay at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:615
--- 0x4201cad4: AfeWakeWord::AudioDetectionTask() at H:/project/7.0inch/ESP32/7.0inch_ESP32/newest_xiaozhi_AI/xiaozhi-esp32-800x480/main/audio_processing/afe_wake_word.cc:122
--- 0x4201cb61: AfeWakeWord::Initialize(AudioCodec*)::{lambda(void*)#1}::_FUN(void*) at H:/project/7.0inch/ESP32/7.0inch_ESP32/newest_xiaozhi_AI/xiaozhi-esp32-800x480/main/audio_processing/afe_wake_word.cc:75
---  (inlined by) _FUN at H:/project/7.0inch/ESP32/7.0inch_ESP32/newest_xiaozhi_AI/xiaozhi-esp32-800x480/main/audio_processing/afe_wake_word.cc:77 

ELF file SHA256: 124d77b6a

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379cfd
--- 0x40379cfd: esp_restart_noos at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/port/soc/esp32s3/system_internal.c:160

=======================================================

第二种： 
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  1 register dump:
PC      : 0x403783c0  PS      : 0x00060f36  A0      : 0x4038b646  A1      : 0x3fcaf5f0
--- 0x403783c0: _DoubleExceptionVector at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/xtensa/xtensa_vectors.S:564
--- 0x4038b646: _xt_context_save at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/xtensa/xtensa_context.S:191

A2      : 0x3fcaf660  A3      : 0x00000000  A4      : 0x803835c6  A5      : 0x3fcc8f30
A6      : 0x00060023  A7      : 0x00000003  A8      : 0x820c7b52  A9      : 0x3fcaf670
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x803875e6  A13     : 0x3fcd75d0
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x00000000  EXCCAUSE: 0x00000001
EXCVADDR: 0x0000000c  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x403783bd:0x3fcaf5f0 0x4038b643:0x3fcaf6d0 0x40384737:0x3fcaf6f0
--- 0x403783bd: _UserExceptionVector at ??:?
--- 0x4038b643: _xt_context_save at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/xtensa/xtensa_context.S:167
--- 0x40384737: prvIdleTask at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/freertos/FreeRTOS-Kernel/tasks.c:4353 (discriminator 1)     

ELF file SHA256: 124d77b6a

Rebooting...
  ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420926c4
--- 0x420926c4: dl_tie728_pointwise_conv1d_qacc_i16o16k8b16_round_ltz_end at /home/sunxiangyu/workspace/esp_sr_lib/components/dl_lib/base/tie728/dl_tie728_pointwise_conv1d_qacc_i16o16k8b16.S:176
[... 80 lines cut ...]
A2      : 0x3fca3ff0  A3      : 0x00000000  A4      : 0xb33fffff  A5      : 0x3fcaf640
A6      : 0x00060123  A7      : 0x00000000  A8      : 0x60023000  A9      : 0x60023000
A10     : 0x00000000  A11     : 0x40000000  A12     : 0x820c7e9a  A13     : 0x3fcaf610
--- 0x40000000: _WindowOverflow4 in ROM

[... cut ...]
--- 0x40383786: _frxt_int_exit at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246   
--- 0x4037c051: _xt_lowint1 at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/xtensa/xtensa_vectors.S:1233

ELF file SHA256: 124d77b6a

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379cfd
--- 0x40379cfd: esp_restart_noos at D:/ESP32/Espressif/frameworks/esp-idf-v5.4.1/components/esp_system/port/soc/esp32s3/system_internal.c:160