78/xiaozhi-esp32 #872: 新建iot后无法识别到新建的iot

lixd860059149, 2025-07-01:
### Is your feature request related to a problem?

新做一个iot获取设备温湿度后一直无法识别到是什么原因，我用的是面包板

[image]
具体代码实现如下

[image]

[image]
日志如下

[image]

78, 2025-07-01:
请参考 MCP 使用方法：https://github.com/78/xiaozhi-esp32/blob/main/docs/mcp-usage.md

lixd860059149, 2025-07-02:
> 请参考 MCP 使用方法：https://github.com/78/xiaozhi-esp32/blob/main/docs/mcp-usage.md

以后是准备废弃掉iot这块要改成mcp方式么，还有目前我调试iot这块是什么原因呢，我是小白好多地方不懂还望大神指点

lixd860059149, 2025-07-02:
> 请参考 MCP 使用方法：https://github.com/78/xiaozhi-esp32/blob/main/docs/mcp-usage.md

我换成mcp后为啥还是识别不到，麻烦再指点下

[image]

78, 2025-07-02:
可以看运行日志输出来查看是否添加成功。