78/xiaozhi-esp32 #883: xiaozhi-esp32-1.7.2版本编译出现问题

kkang520, 2025-07-03:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

xiaozhi-esp32-1.7.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

请帮我解决这个问题

### What is the actual behavior?

xiaozhi-esp32-1.7.2版本编译出现问题

### Steps to reproduce.

使用powershell编译xiaozhi-esp32-1.7.2版本编译的时候出现的报错

### Build or installation Logs.

```plain
during RTL pass: ira
E:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_lcd/rgb/esp_lcd_panel_rgb.c: In function 'rgb_panel_draw_bitmap':
E:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_lcd/rgb/esp_lcd_panel_rgb.c:681:1: internal compiler error: Segmentation fault
  681 | }
      | ^
```

78, 2025-07-05:
可能是主机内存不足？

XieAndWater, 2025-07-09:
idf.py fullclean
idf.py build