78/xiaozhi-esp32 #884: image recognition fails

RealDeco, 2025-07-03:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

1.76

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

to say what it see

### What is the actual behavior?

saying technical problems and in logs i can see:

{"result":{"success":false,"message":"Failed to upload photo"}}

### Steps to reproduce.

ask what it see, it shows image on screen but say it has problems getting the feed so it can's describe it.

...

### Build or installation Logs.

```plain

```

### More Information.

using waveshare esp32-s3-touch-lcd-3.5

[image]

78, 2025-07-05:
It appears that a network error occurred during the image explanation. We will address this issue next week.

RealDeco, 2025-07-05:
no hurry i'm just testing and since it worked before i thought i better say something.
Great work! Imagine what a portable voice assist with image gen like this could do for blind people! :)