78/xiaozhi-esp32 #917: Unable to turn on device due to incorrect source code loaded

hiepeng, 2025-07-14:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

esp32c3-api1-20210207

### Operating System used.

macOS

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

[image]

ESP-ROM:esp32c3-api1-20210207

I bought a device online, and I came back and reloaded the source code, but I don't know which board is correct, I tried reloading the source code for many different boards, but none of them can turn on, please help me.

[image]

[image]

[image]

[image]

[image]

### What is the actual behavior?

[image]

ESP-ROM:esp32c3-api1-20210207

I bought a device online, and I came back and reloaded the source code, but I don't know which board is correct, I tried reloading the source code for many different boards, but none of them can turn on, please help me.

[image]

[image]

[image]

[image]

[image]

### Steps to reproduce.

[image]

ESP-ROM:esp32c3-api1-20210207

I bought a device online, and I came back and reloaded the source code, but I don't know which board is correct, I tried reloading the source code for many different boards, but none of them can turn on, please help me.

[image]

[image]

[image]

[image]

[image]

### Build or installation Logs.

```plain

```

HonestQiao, 2025-07-15:
For purchased Xiao Zhi devices, if you want to perform secondary development or compile/flash firmware yourself, you must do the following:

Boot the device with its original firmware, then check the SKU in the output logs. For example: SKU=xmini-c3.

Even after obtaining the SKU (e.g., SKU=xmini-c3), it may not be sufficient for secondary development based solely on the original code. It’s best to ask the seller for the schematic diagram or request specific pin definitions.

You should now contact the seller to obtain the original firmware. Ideally, also request the schematic diagram.

hiepeng, 2025-07-15:
esp32c3

closed 2025-07-15 by hiepeng