78/xiaozhi-esp32 #945: esp32p4 sdk configuration 报错

Forone-Shijie, 2025-07-19:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.8.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

我在用vscode esp-idf 插件运行小智源代码的sdk configuration 时，我选择开发板为esp32p4时，就会出现错误，报错如下：
---------------------------ERROR--------------------------

info: INFO: Symbol ESP_IDF_VERSION defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:

    F:\Project\xiaozhi\managed_components\espressif__i2c_bus\Kconfig:4

    F:\Project\xiaozhi\managed_components\espressif__esp_codec_dev\Kconfig:2

-----------------------END OF ERROR-----------------------
Server running, waiting for requests on stdin...

---------------------------ERROR--------------------------

Traceback (most recent call last):
  File "asyncio\streams.py", line 545, in readline
  File "asyncio\streams.py", line 640, in readuntil
asyncio.exceptions.LimitOverrunError: Separator is found, but chunk is longer than limit

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\esp-idf\tools\idf.py", line 855, in <module>
    main()
  File "D:\esp-idf\tools\idf.py", line 745, in main
    cli(argv, prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 1161, in __call__

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 1729, in invoke
    return _process_result(rv)
           ^^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 1666, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Espressif\python_env\idf5.4_py3.11_env\Lib\site-packages\click\core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\esp-idf\tools\idf.py", line 640, in execute_tasks

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    task(ctx, global_args, task.action_args)
  File "D:\esp-idf\tools\idf.py", line 212, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "D:\esp-idf\tools\idf_py_actions\core_ext.py", line 50, in build_target
    run_target(target_name, args, force_progression=GENERATORS[args.generator].get('force_progression', False))
  File "D:\esp-idf\tools\idf_py_actions\tools.py", line 499, in run_target
    RunTool(generator_cmd[0], generator_cmd + [target_name], args.build_dir, env, custom_error_handler, hints=not args.no_hints,
  File "D:\esp-idf\tools\idf_py_actions\tools.py", line 319, in __call__
    process, stderr_output_file, stdout_output_file = asyncio.run(self.run_command(self.args, env_copy))
                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "asyncio\runners.py", line 190, in run

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

  File "asyncio\runners.py", line 118, in run
  File "asyncio\base_events.py", line 653, in run_until_complete
  File "D:\esp-idf\tools\idf_py_actions\tools.py", line 364, in run_command
    await asyncio.gather(
  File "D:\esp-idf\tools\idf_py_actions\tools.py", line 441, in read_and_write_stream
    output = await read_stream()
             ^^^^^^^^^^^^^^^^^^^
  File "D:\esp-idf\tools\idf_py_actions\tools.py", line 399, in read_stream

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    output_b = await input_stream.readline()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "asyncio\streams.py", line 554, in readline
ValueError: Separator is found, but chunk is longer than limit

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

Exception ignored in: <function BaseSubprocessTransport.__del__ at 0x0000025AE6F3C180>
Traceback (most recent call last):
  File "asyncio\base_subprocess.py", line 125, in __del__

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

  File "asyncio\base_subprocess.py", line 78, in __repr__
  File "asyncio\proactor_events.py", line 80, in __repr__
  File "asyncio\windows_utils.py", line 102, in fileno
ValueError: I/O operation on closed pipe
Exception ignored in: <function _ProactorBasePipeTransport.__del__ at 0x0000025AE6F3D800>
Traceback (most recent call last):
  File "asyncio\proactor_events.py", line 116, in __del__
  File "asyncio\proactor_events.py", line 80, in __repr__
  File "asyncio\windows_utils.py", line 102, in fileno
ValueError: I/O operation on closed pipe

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 1
-----------------------END OF ERROR-----------------------
问了gemini，说是什么路径长度问题，但是修改到比较短后仍然存在；
但是将开发板改为esp32s3后，该错误就不存在，请问这个是什么问题，应该如何解决？

另外，选p4的时候代码build没有出错，我的esp安装位置为：
D:\esp-idf
D:\Espressif
安装版本是v5.4

### What is the actual behavior?

配置esp32p4开发板的sdk configuration时报错，核心错误为：
Traceback (most recent call last):
  File "asyncio\streams.py", line 545, in readline
  File "asyncio\streams.py", line 640, in readuntil
asyncio.exceptions.LimitOverrunError: Separator is found, but chunk is longer than limit

### Steps to reproduce.

1. 打开vscode
2. 选择esp-idf
3. 选择esp32p4开发板
4. 点击sdk configuration
...

### Build or installation Logs.

```plain

```

78, 2025-07-19:
用 Linux 和 MacOS 问题比较少

closed 2025-07-20 by Forone-Shijie

zjy212321, 2025-07-30:
> 用 Linux 和 MacOS 问题比较少

我和这位的问题一模一样, 我的系统是macos