78/xiaozhi-esp32 #956: 音频代码AFE 的format 是不是错的

xundanqing, 2025-07-22:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

main

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

[image]
如果接入为两个mic 和一路回采信号不应该是 MMR 或者RMNM 这种吗？怎么回事MR 呢

### What is the actual behavior?

如果接入为两个mic 和一路回采信号不应该是 MMR 或者RMNM 这种吗？怎么回事MR 呢

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

xundanqing, 2025-08-01:
硬件参数：es7210  mic1，mic2--接mic  mi3 接回采信号
如果ES7210 设置4个通道，16000/ 4mic  /16bit  ESP32 I2S 设置为16000 4ch 16bit
i2s 读取到的数据格式为MRMN(1324)  ,如果esp32  i2s设置为32bit .2通道，读取到的数据格式为RMNM (3142)

Rockyaaaaa, 2025-09-19:
如果 afe_type_t 的参数选择 AFE_TYPE_VC，并且设置了"MRM"，初始化的日志会提示：

> W (6367) AFE_CONFIG: AFE_TYPE_VC only support single microphone channel. If input is multi-channels, the first channel will be selected
> W (6367) AFE_CONFIG: For single microphone channel, SE is deactivated.

如果是 AFE_TYPE_SR，则不会有这个报警