78/xiaozhi-esp32 #957: bug

lr-123456, 2025-07-22:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

1.8版本

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

内存没有溢出应该正常运行把

### What is the actual behavior?

新代码烧录后为什么总重启

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

wuhuzi, 2025-07-22:
same issue

lr-123456, 2025-07-22:
我是1.8.0正常烧录没问题，但是改成自定义唤醒词烧入就一直重启

78, 2025-07-22:
可以提供报错信息

ZhouKes, 2025-07-22:
@lr-123456 @lamnickdavid 看看报错信息

lr-123456, 2025-07-22:
@ZhouKes 没有报错信息，就是1.8.0版本自定义唤醒词，编译正常，但烧录进去后一直重新联网，1.8.1正常编译烧录都重启，个人猜测可能是初始化时候内存的问题或是Windows系统的编码问题吧

ZhouKes, 2025-07-22:
@lr-123456 立创S3刚试过1.8.1，没有问题。如果是windows系统编码问题，在编译的时候就会报错。

lamnickdavid, 2025-07-22:
> @lr-123456 @lamnickdavid 看看报错信息

有没有可能是16m_custom_wakeword.csv的问题