78/xiaozhi-esp32 #975: 新版本自定唤醒词，不重启了，但没有反应，根本换不醒

lr-123456, 2025-07-24:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

1.8.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

预期，更改唤醒词后，可以正常唤醒

### What is the actual behavior?

更改唤醒词后，没有反应，我的板子是面包版+lcd+cam

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

sfdsl, 2025-07-24:
遇到同样的问题，

sfdsl, 2025-07-24:
刚刚试了下，先把整改项目清除编译idf.py clean，再次编译idf.py build，然后下载；就可以用新的唤醒词，唤醒设备了。

closed 2025-07-24 by lr-123456