78/xiaozhi-esp32 #983: 通过4G模组（ML307）访问Xiaozhi-esp32-server时，WSS握手失败

ZidongY, 2025-07-25:
我手里有3个小智AI开发板：
1. 无名_1.54_ML307，固件版本：1.7.3
2. 斑梨_1.28 LCD_ML307，固件版本：1.6.8
3. 立创_实战派_C3，固件版本：1.8.2

使用[xiaozhi-esp32-server](https://github.com/xinnan-tech/xiaozhi-esp32-server/tree/main)项目部署了一套小智服务后台，并用Ningx配置了SSL证书做了HTTPS和WSS的反向代理。

服务后台：
管理系统：[https://aitoy.hdtron.com:8902/](https://aitoy.hdtron.com:8902/)
OTA地址：[https://aitoy.hdtron.com:8902/xiaozhi/ota/](https://aitoy.hdtron.com:8902/xiaozhi/ota/)

问题：
1. 当使用“无名_1.54_ML307”和“立创_实战派_C3”开发板通过Wifi访问后台时，一切正常。日志可以清晰看到WSS验证成功。如截图

[image]

2. 当使用4G访问后台时，可以登录平台通过设备认证，但对话一启动，就直接报错WSS握手失败。如截图

[image]

3. 当我怀疑是4G模组问题时，我又将设备连接了xiaozhi-esp32-server的测试平台，发现WSS连接又是成功的。但跟管理员沟通后，发现Nginx配置上并没啥本质差别。
```
OTA接口地址: https://2662r3426b.vicp.fun/xiaozhi/ota/
设备管理地址: https://2662r3426b.vicp.fun/
```

由于固件开发这边我并不是很熟悉，所以希望有能力的朋友帮忙看下这个问题。
这个问题我已经排查了2天了，但还是没有具体头绪，现在只能有个“大胆猜测”：难道ML307模组建链WSS连接时，只会限定标准的443端口，非标端口都拒绝连接吗？o(╥﹏╥)o

openrz, 2025-07-25:
我手头上有一台4G，但是能连接你的OTA接口

[image]

[image]

78, 2025-07-25:
老版本可能只支持443的wss，建议升级到最新版本

ZidongY, 2025-07-25:
> 我手头上有一台4G，但是能连接你的OTA接口
> 
> 感觉像是python的8000端口没启动
> 
> <img alt="Image" width="883" height="392" src="https://private-user-images.githubusercontent.com/32005838/470622632-16c819f0-c3ac-42d3-ae69-a22a38bda144.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM0MzAzODksIm5iZiI6MTc1MzQzMDA4OSwicGF0aCI6Ii8zMjAwNTgzOC80NzA2MjI2MzItMTZjODE5ZjAtYzNhYy00MmQzLWFlNjktYTIyYTM4YmRhMTQ0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzI1VDA3NTQ0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTllODlkMDM3MzQ2OWVlOWY1NGQ5MDQ4N2EyY2ZhZjA2MjFiN2I3ZWViNjZlNzZjNDlhZDVlN2YwNzNkNWM1OTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.TCOnj09PDZGtyHeZQz5o1zGc7990opYjT2ziR_4FmTY">
> [image]

服务端的8000端口是通的，下面是测试截图。另外，如果不同的话，wifi访问最后也是代理到8000端口，应该无法对话才对。
[image]

78, 2025-07-25:
是需要更新 esp-ml307 组件

ZidongY, 2025-07-31:
经过一番摸索，找到了一种临时解决方案。
由于厂家提供的源码比较老， esp-ml307 组件版本还是2.1.x，我已经尝试更新到了3.2.5，但需要配套修改大量的固件源码，否则固件编译会报错。
对此我找到了另一种简单的解决方案，直接修改esp-ml307组件2.x版本的源码。
通过问题定位，需要修改的文件是：`.\managed_components\78__esp-ml307\ml307_ssl_transport.cc`

问题代码在93行附近，如下图所示。
[image]

通过源码可以看出，问题的产生是为了解决老版本4G模组对话必须通过WSS访问的问题，所以增加了一个判断语句遇到443端口的WS URL地址，执行SSL保护，否则通过普通WS建立连接。
就是因为这个改动，导致我这的情况被误伤了~~ 
为了不改动后台配置，这里我把4G模块的访问还原为强制WSS建链，具体修改如下。

```ml307_ssl_transport.cc
    // 只对 443 端口使用 SSL
    //sprintf(command, "AT+MIPCFG=\"ssl\",%d,%d,0", tcp_id_, port == 443 ? 1 : 0);
    sprintf(command, "AT+MIPCFG=\"ssl\",%d,1,0", tcp_id_);
    if (!modem_.Command(command)) {
        ESP_LOGE(TAG, "Failed to set TCP SSL configuration");
        return false;
    }
```

78, 2025-07-31:
是的，现在会根据 url 里是 wss 还是 ws 来决定是否使用 ssl，你用的组件版本应该是比较老了。