78/xiaozhi-esp32 #991: 1.8.0+新版本 编译运行遇到了SSL链接和I2S音频通道报错

PanYFeng, 2025-07-27:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

1.8.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

I (5880) Ota: Current version: 1.8.2
I (13110) esp-x509-crt-bundle: Certificate validated
E (13640) Dynamic Impl: mbedtls_ssl_fetch_input error=29312
I (13650) Ota: Current is the latest version
I (13650) Ota: Running partition: ota_0
I (13660) MCP: Add tool: self.get_device_status
I (13660) MCP: Add tool: self.audio_speaker.set_volume
I (13660) MCP: Add tool: self.screen.set_brightness
I (13660) MCP: Add tool: self.screen.set_theme
I (13660) MQTT: Connecting to endpoint mqtt.xiaozhi.me
I (15650) AudioCodec: Set input enable to false
I (15660) AudioCodec: Set output enable to false
I (15660) SystemInfo: free sram: 94131 minimal sram: 90319
I (20020) esp-x509-crt-bundle: Certificate validated
I (20380) MQTT: Connected to endpoint
I (20380) Application: STATE: idle
I (20390) MODEL_LOADER: The storage free size is 20608 KB
I (20390) MODEL_LOADER: The partition size is 960 KB
I (20390) MODEL_LOADER: Successfully load srmodels
I (20390) AfeWakeWord: Model 0: wn9_nihaoxiaozhi_tts
I (20390) AFE_CONFIG: Set WakeNet Model: wn9_nihaoxiaozhi_tts
MC Quantized wakenet9: wakenet9l_tts1h8_你好小智_3_0.631_0.635, tigger:v4, mode:0, p:0, (Jun 16 2025 16:18:15)
I (20490) AFE: AFE Version: (1MIC_V250121)
I (20490) AFE: Input PCM Config: total 2 channels(1 microphone, 1 playback), sample rate:16000
I (20500) AFE: AFE Pipeline: [input] -> |AEC(SR_HIGH_PERF)| ->  -> |VAD(WebRTC)| -> |WakeNet(wn9_nihaoxiaozhi_tts,)| -> [output]
I (20510) AfeWakeWord: Audio detection task started, feed size: 512 fetch size: 512
E (20520) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet
I (20520) I2S_IF: channel mode 2 bits:16/16 channel:4 mask:3
I (20520) AudioService: Resampling audio from 16000 to 24000
I (20530) I2S_IF: TDM Mode 0 bits:16/16 channel:4 sample_rate:24000 mask:3
I (20530) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000
E (20540) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet
E (20560) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet
I (20560) I2S_IF: channel mode 0 bits:16/16 channel:4 mask:3
I (20560) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
I (20570) I2S_IF: STD Mode 1 bits:32/32 channel:4 sample_rate:24000 mask:0
I (20580) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:24000 mask:1
I (20580) ES7210: Bits 16
E (20590) i2s_common: i2s_channel_enable(1165): the channel has already enabled or not initialized
I (20600) ES7210: Enable ES7210_INPUT_MIC1
I (20520) SystemInfo: free sram: 68251 minimal sram: 67427
I (20610) ES7210: Enable ES7210_INPUT_MIC2
I (20620) ES7210: Enable ES7210_INPUT_MIC3
I (20630) ES7210: Enable ES7210_INPUT_MIC4
I (20630) ES7210: Enable TDM mode
I (20630) Adev_Codec: Open codec device OK
I (20630) AudioCodec: Set output enable to true
I (20640) ES7210: Unmuted
I (20640) Adev_Codec: Open codec device OK
I (20640) AudioCodec: Set input enable to true
--- 0x4200bdb2: BoxAudioCodec::Read(short*, int) at D:/esp32/OpenSource/xiaozhi-esp32-1.8.2/main/audio/codecs/box_audio_codec.cc:232 (discriminator 1)
---  (inlined by) BoxAudioCodec::Read(short*, int) at D:/esp32/OpenSource/xiaozhi-esp32-1.8.2/main/audio/codecs/box_audio_codec.cc:230 (discriminator 1)

dioCodec: Set input enable to false

file: "./main/audio/codecs/box_audio_codec.cc" line 232
func: virtual int BoxAudioCodec::Read(int16_t*, int)
expression: esp_codec_dev_read(input_dev_, (void*)dest, samples * sizeof(int16_t))
E (21660) i2s_common: i2s_channel_disable(1191): the channel has not been enabled y

### What is the actual behavior?

日记报告错误

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

78, 2025-07-27:
看起来没有影响使用，可以尝试使用最新版的 ESP-IDF，看看是否还存在报错。

PanYFeng, 2025-07-27:
我现在用的是ESP-IDF V5.4.1 应该是最新版了吧？ 我是基于立创实战派s3，去掉了扩展芯片，加入了AXP电源芯片，重新设计的一块版， 请问下其它板子有没有这个报错，有的话，就等着老哥修复了，不然我得找我自己的问题。。。有时候会重启一两次。

78, 2025-07-28:
把 Audio Service 的低功耗定时器禁用试试，不会是有硬件故障吧？

PanYFeng, 2025-07-29:
void AudioService::CheckAndUpdateAudioPowerState() {
    /*
    auto now = std::chrono::steady_clock::now();
    auto input_elapsed = std::chrono::duration_cast<std::chrono::milliseconds>(now - last_input_time_).count();
    auto output_elapsed = std::chrono::duration_cast<std::chrono::milliseconds>(now - last_output_time_).count();
    if (input_elapsed > AUDIO_POWER_TIMEOUT_MS && codec_->input_enabled()) {
        codec_->EnableInput(false);
    }
    if (output_elapsed > AUDIO_POWER_TIMEOUT_MS && codec_->output_enabled()) {
        codec_->EnableOutput(false);
    }
    if (!codec_->input_enabled() && !codec_->output_enabled()) {
        esp_timer_stop(audio_power_timer_);
    }
    */
}
十分感谢解答！
注释掉这段代码后没有报错
硬件原理基本照搬立创S3加了电源芯片，除了 I2C USB  SPI FLASH  UART  GPIO保持没变，其它IO为了布线方便，调整对应的IO口，
是硬件上的差异导致吗
软件上能否解决，重构音频代码前没有报这个错误的，
我看日记有 多次I2S通道开关，能否强化下I2S 通道状态及切换逻辑
I (378) LcdDisplay: Adding LCD display
I (418) Backlight: Set brightness to 75
I (418) Application: STATE: starting
D (418) i2s_common: tx channel is registered on I2S0 successfully
D (418) i2s_common: rx channel is registered on I2S0 successfully
D (428) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 960
D (438) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 8 [bclk] 768000 Hz
D (448) i2s_common: MCLK is pinned to GPIO41 on I2S0
D (448) i2s_std: The tx channel on I2S0 has been initialized to STD mode successfully
D (458) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 1920
D (468) i2s_tdm: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
D (478) i2s_common: MCLK is pinned to GPIO41 on I2S0
D (478) i2s_tdm: The rx channel on I2S0 has been initialized to TDM mode successfully
I (488) BoxAudioCodec: Duplex channels created
I (498) ES8311: Work in Slave mode
I (508) ES7210: Work in Slave mode
I (508) ES7210: Enable ES7210_INPUT_MIC1
I (518) ES7210: Enable ES7210_INPUT_MIC2
I (518) ES7210: Enable ES7210_INPUT_MIC3
I (518) ES7210: Enable ES7210_INPUT_MIC4
I (528) ES7210: Enable TDM mode
I (528) BoxAudioCodec: BoxAudioDevice initialized
D (528) i2s_common: i2s tx channel enabled
D (528) i2s_common: i2s rx channel enabled
D (528) i2s_common: i2s rx channel disabled
I (538) I2S_IF: channel mode 2 bits:16/16 channel:4 mask:3
D (538) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 960
D (548) i2s_tdm: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
I (558) I2S_IF: TDM Mode 0 bits:16/16 channel:4 sample_rate:24000 mask:3
D (568) i2s_common: i2s tx channel disabled
I (568) I2S_IF: channel mode 0 bits:16/16 channel:4 mask:3
D (578) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 1920
D (588) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
I (598) I2S_IF: STD Mode 1 bits:32/32 channel:4 sample_rate:24000 mask:0
D (608) i2s_common: i2s tx channel enabled
D (608) i2s_common: i2s rx channel enabled
I (608) ES7210: Bits 16
I (618) ES7210: Enable ES7210_INPUT_MIC1
I (628) ES7210: Enable ES7210_INPUT_MIC2
I (628) ES7210: Enable ES7210_INPUT_MIC3
I (628) ES7210: Enable ES7210_INPUT_MIC4
I (628) ES7210: Enable TDM mode
I (638) ES7210: Unmuted
I (638) Adev_Codec: Open codec device OK
I (638) AudioCodec: Set input enable to true
D (638) i2s_common: i2s tx channel disabled
I (648) I2S_IF: Mode 0 need extend bits 64 to 32
I (648) I2S_IF: channel mode 0 bits:16/32 channel:2 mask:1
D (658) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (668) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
I (678) I2S_IF: STD Mode 1 bits:16/32 channel:2 sample_rate:24000 mask:1
D (678) i2s_common: i2s tx channel enabled
I (698) Adev_Codec: Open codec device OK
I (698) AudioCodec: Set output enable to true
I (698) AudioCodec: Audio codec started

78, 2025-07-29:
你可以单独注释 `codec_->EnableInput(false);` 看看是否关闭 mic 引起，还是关闭 speaker 引起。

whni, 2025-07-29:
@78 我也是是使用立创s3开发板，没有任何的改动，基本上可以确认是在这个commit之后这个问题出现的：https://github.com/quebotics/quechat-esp32/commit/679012c2762e94bb395bedcdff2c6775492a0c8b

PanYFeng, 2025-07-29:
> 你可以单独注释 `codec_->EnableInput(false);` 看看是否关闭 mic 引起，还是关闭 speaker 引起。
以下是注释 `codec_->EnableInput(false);` 日记 没有报错
I (388) s3 ll_cam: DMA Channel=1
I (388) cam_hal: cam init ok
E (428) camera: Detected camera not supported.
E (428) camera: Camera probe failed with error 0x106(ESP_ERR_NOT_SUPPORTED)
E (428) gdma: gdma_disconnect(309): no peripheral is connected to the channel
I (438) PowerSaveTimer: Power save timer enabled
I (438) Backlight: Set brightness to 75
I (448) Application: STATE: starting
D (448) i2s_common: tx channel is registered on I2S0 successfully
D (448) i2s_common: rx channel is registered on I2S0 successfully
D (458) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 960
D (468) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 8 [bclk] 768000 Hz
D (478) i2s_common: MCLK is pinned to GPIO41 on I2S0
D (488) i2s_std: The tx channel on I2S0 has been initialized to STD mode successfully
D (488) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 1920
D (498) i2s_tdm: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
D (508) i2s_common: MCLK is pinned to GPIO41 on I2S0
D (518) i2s_tdm: The rx channel on I2S0 has been initialized to TDM mode successfully
I (528) BoxAudioCodec: Duplex channels created
I (538) ES8311: Work in Slave mode
I (538) ES7210: Work in Slave mode
I (548) ES7210: Enable ES7210_INPUT_MIC1
I (548) ES7210: Enable ES7210_INPUT_MIC2
I (548) ES7210: Enable ES7210_INPUT_MIC3
I (558) ES7210: Enable ES7210_INPUT_MIC4
I (558) ES7210: Enable TDM mode
I (558) BoxAudioCodec: BoxAudioDevice initialized
D (558) i2s_common: i2s tx channel enabled
D (558) i2s_common: i2s rx channel enabled
D (568) i2s_common: i2s rx channel disabled
I (568) I2S_IF: channel mode 2 bits:16/16 channel:4 mask:3
D (578) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 960
D (588) i2s_tdm: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
I (598) I2S_IF: TDM Mode 0 bits:16/16 channel:4 sample_rate:24000 mask:3
D (598) i2s_common: i2s tx channel disabled
I (608) I2S_IF: channel mode 0 bits:16/16 channel:4 mask:3
D (608) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 1920
D (618) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
I (628) I2S_IF: STD Mode 1 bits:32/32 channel:4 sample_rate:24000 mask:0
D (638) i2s_common: i2s tx channel enabled
D (638) i2s_common: i2s rx channel enabled
I (648) ES7210: Bits 16
I (658) ES7210: Enable ES7210_INPUT_MIC1
I (658) ES7210: Enable ES7210_INPUT_MIC2
I (658) ES7210: Enable ES7210_INPUT_MIC3
I (668) ES7210: Enable ES7210_INPUT_MIC4
I (668) ES7210: Enable TDM mode
I (668) ES7210: Unmuted
I (668) Adev_Codec: Open codec device OK
I (678) AudioCodec: Set input enable to true
D (678) i2s_common: i2s tx channel disabled
I (678) I2S_IF: Mode 0 need extend bits 64 to 32
I (688) I2S_IF: channel mode 0 bits:16/32 channel:2 mask:1
D (688) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (698) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
I (708) I2S_IF: STD Mode 1 bits:16/32 channel:2 sample_rate:24000 mask:1
D (718) i2s_common: i2s tx channel enabled
I (738) Adev_Codec: Open codec device OK
I (738) AudioCodec: Set output enable to true
I (738) AudioCodec: Audio codec started
I (738) OpusResampler: Resampler configured with input sample rate 24000 and output sample rate 16000
I (748) OpusResampler: Resampler configured with input sample rate 24000 and output sample rate 16000
。。。。。。。。。。
ask: 255.255.255.0, gw: 192.168.8.1
I (5758) Application: STATE: activating
I (5768) Ota: Current version: 1.8.3
。。。。。。。。。
I (13458) MQTT: Connecting to endpoint mqtt.xiaozhi.me
[... 76 lines cut ...]
I (202098) Application: STATE: listening
I (211758) SystemInfo: free sram: 51343 minimal sram: 49955
I (217758) I2S_IF: Pending out channel for in channel running
I (217758) AudioCodec: Set output enable to false
I (221758) SystemInfo: free sram: 50763 minimal sram: 49887
I (231758) SystemInfo: free sram: 51343 minimal sram: 49887
I (241758) SystemInfo: free sram: 50827 minimal sram: 49887
I (251758) SystemInfo: free sram: 51343 minimal sram: 49887
I (261758) SystemInfo: free sram: 51279 minimal sram: 49887
I (268288) Application: STATE: speaking
I (268908) Application: << 那我先休息啦，有需要随时喊我~
D (269128) i2s_common: i2s tx channel disabled
I (269128) I2S_IF: Mode 0 need extend bits 64 to 32
I (269128) I2S_IF: channel mode 0 bits:16/32 channel:2 mask:1
D (269128) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
I (269138) I2S_IF: STD Mode 1 bits:16/32 channel:2 sample_rate:24000 mask:1
D (269148) i2s_common: i2s tx channel enabled
I (269188) Adev_Codec: Open codec device OK
I (269188) AudioCodec: Set output enable to true
I (271708) Application: STATE: listening
I (271918) MQTT: Received goodbye message, session_id: f66116ef
I (271928) wifi:Set ps type: 1, coexist: 0

I (271938) Application: STATE: idle
I (281758) SystemInfo: free sram: 54731 minimal sram: 49883

PanYFeng, 2025-07-29:
注释//codec_->EnableOutput(false);这一句，有一处报错，语音唤醒 声音均正常使用

I (13528) MQTT: Connecting to endpoint mqtt.xiaozhi.me
D (15758) i2s_common: i2s rx channel disabled
I (15758) AudioCodec: Set input enable to false
I (15758) SystemInfo: free sram: 92611 minimal sram: 92111
I (20078) esp-x509-crt-bundle: Certificate validated
I (20438) MQTT: Connected to endpoint
I (20438) Application: STATE: idle
I (20448) MODEL_LOADER: The storage free size is 20608 KB
I (20448) MODEL_LOADER: The partition size is 960 KB
I (20448) MODEL_LOADER: Successfully load srmodels
I (20448) AfeWakeWord: Model 0: wn9_nihaoxiaozhi_tts
I (20458) AFE_CONFIG: Set WakeNet Model: wn9_nihaoxiaozhi_tts
MC Quantized wakenet9: wakenet9l_tts1h8_你好小智_3_0.631_0.635, tigger:v4, mode:0, p:0, (Jun 16 2025 16:18:15)
I (20548) AFE: AFE Version: (1MIC_V250121)
I (20558) AFE: Input PCM Config: total 2 channels(1 microphone, 1 playback), sample rate:16000
I (20558) AFE: AFE Pipeline: [input] -> |AEC(SR_HIGH_PERF)| ->  -> |VAD(WebRTC)| -> |WakeNet(wn9_nihaoxiaozhi_tts,)| -> [output]
I (20578) AfeWakeWord: Audio detection task started, feed size: 512 fetch size: 512
E (20578) i2s_common: i2s_channel_disable(1218): the channel has not been enabled yet
I (20588) I2S_IF: channel mode 2 bits:16/16 channel:4 mask:3
I (20588) AudioService: Resampling audio from 16000 to 24000
D (20598) i2s_tdm: Clock division info: [sclk] 160000000 Hz [mdiv] 26 1/24 [mclk] 6144000 Hz [bdiv] 4 [bclk] 1536000 Hz
I (20598) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000
I (20608) I2S_IF: TDM Mode 0 bits:16/16 channel:4 sample_rate:24000 mask:3
D (20628) i2s_common: i2s rx channel enabled
I (20628) ES7210: Bits 16
I (20588) SystemInfo: free sram: 66763 minimal sram: 65935
I (20638) ES7210: Enable ES7210_INPUT_MIC1
I (20648) ES7210: Enable ES7210_INPUT_MIC2
I (20648) ES7210: Enable ES7210_INPUT_MIC3
I (20648) ES7210: Enable ES7210_INPUT_MIC4
I (20658) ES7210: Enable TDM mode
I (20658) ES7210: Unmuted
I (20658) Adev_Codec: Open codec device OK
I (20658) AudioCodec: Set input enable to true
I (29758) SystemInfo: free sram: 67891 minimal sram: 63611
I (35808) Application: STATE: connecting
I (35848) MQTT: Session ID: 50c370aa
I (35858) wifi:Set ps type: 0, coexist: 0

I (35858) Application: Wake word detected: 你好小智
I (36238) AfeWakeWord: Encode wake word opus 66 packets in 437 ms
I (36248) Application: STATE: listening
I (36278) AFE: AFE Version: (1MIC_V250121)
I (36278) AFE: Input PCM Config: total 2 channels(1 microphone, 1 playback), sample rate:16000
I (36278) AFE: AFE Pipeline: [input] -> |AEC(VOIP_HIGH_PERF)| -> [output]
I (36288) AfeAudioProcessor: Audio communication task started, feed size: 256 fetch size: 512
I (36298) Application: >> Hi, 小智
I (36298) Application: STATE: speaking
I (36958) Application: << 嗨嗨，我是小智，有什么可以帮你的吗？
I (40118) Application: STATE: listening
I (49758) SystemInfo: free sram: 52283 minimal sram: 50471
I (59758) SystemInfo: free sram: 51971 minimal sram: 50471
I (69758) SystemInfo: free sram: 52487 minimal sram: 50471
I (79758) SystemInfo: free sram: 51971 minimal sram: 50471
I (89758) SystemInfo: free sram: 51971 minimal sram: 50471
I (99758) SystemInfo: free sram: 51971 minimal sram: 50471
I (108748) Application: STATE: speaking
I (109358) Application: << 那我先退下啦，有需要随时喊我！
I (112158) Application: STATE: listening
I (112358) MQTT: Received goodbye message, session_id: 50c370aa
I (112368) wifi:Set ps type: 1, coexist: 0

I (112388) Application: STATE: idle
I (121758) SystemInfo: free sram: 54103 minimal sram: 50383
I (124858) Application: STATE: connecting
I (124908) MQTT: Session ID: a5d2fb0a
I (124908) wifi:Set ps type: 0, coexist: 0

I (124908) Application: Wake word detected: 你好小智
I (125308) AfeWakeWord: Encode wake word opus 66 packets in 449 ms
I (125308) Application: STATE: listening
I (125348) Application: STATE: speaking
I (125358) Application: >> Hi, 小智
I (125868) Application: << 哈喽！
I (127108) Application: << 我是小智，有什么可以帮你的吗？
I (130468) Application: STATE: listening
I (136018) Application: STATE: speaking
I (136018) Application: >> 再见。
I (136468) Application: << 好的，下次见啦！
I (138648) MQTT: Received goodbye message, session_id: a5d2fb0a
I (138648) Application: STATE: listening
I (138668) wifi:Set ps type: 1, coexist: 0

I (138668) Application: STATE: idle

jianglianweilai, 2025-08-13:
> 我尝试精简板级代码排除其它因数，和跟其它的板子进行比较，就是解决不了，如果虾哥有空还请不吝赐教，我是用的VSCODE编译我是把板子信息，分别添加CMakeLists.txt/Kconfig.projbuild选择自己的板子编译的  ESP-IDF 还是V5.4.1版本 （我更新到V5.5试试） 报错的日记（完整的和精简后的报错内容差不多，有时会重启是我尝试修改时导致） I (15560) AudioCodec: Set input enable to false I (15560) AudioCodec: Set output enable to false I (15560) SystemInfo: free sram: 121043 minimal sram: 120555 I (20070) esp-x509-crt-bundle: Certificate validated I (20430) MQTT: Connected to endpoint I (20430) Application: STATE: idle I (20430) MODEL_LOADER: The storage free size is 20736 KB I (20430) MODEL_LOADER: The partition size is 960 KB I (20430) MODEL_LOADER: Successfully load srmodels I (20440) AfeWakeWord: Model 0: wn9_nihaoxiaozhi_tts I (20440) AFE_CONFIG: Set WakeNet Model: wn9_nihaoxiaozhi_tts MC Quantized wakenet9: wakenet9l_tts1h8_你好小智_3_0.631_0.635, tigger:v4, mode:0, p:0, (Jun 16 2025 16:18:15) I (20520) AFE: AFE Version: (1MIC_V250121) I (20520) AFE: Input PCM Config: total 2 channels(1 microphone, 1 playback), sample rate:16000 I (20530) AFE: AFE Pipeline: [input] -> |AEC(SR_HIGH_PERF)| -> -> |VAD(WebRTC)| -> |WakeNet(wn9_nihaoxiaozhi_tts,)| -> [output] I (20540) AfeWakeWord: Audio detection task started, feed size: 512 fetch size: 512 E (20550) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet I (20560) I2S_IF: channel mode 2 bits:16/16 channel:4 mask:3 I (20550) AudioService: Resampling audio from 16000 to 24000 I (20560) I2S_IF: TDM Mode 0 bits:16/16 channel:4 sample_rate:24000 mask:3 I (20570) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000 E (20570) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet E (20590) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet I (20600) I2S_IF: channel mode 0 bits:16/16 channel:4 mask:3 I (20600) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1 I (20600) I2S_IF: STD Mode 1 bits:32/32 channel:4 sample_rate:24000 mask:0 I (20610) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:24000 mask:1 I (20620) ES7210: Bits 16 E (20620) i2s_common: i2s_channel_enable(1165): the channel has already enabled or not initialized I (20630) ES7210: Enable ES7210_INPUT_MIC1 I (20570) SystemInfo: free sram: 95155 minimal sram: 94355 I (20640) ES7210: Enable ES7210_INPUT_MIC2 I (20650) ES7210: Enable ES7210_INPUT_MIC3 I (20660) ES7210: Enable ES7210_INPUT_MIC4 I (20660) ES7210: Enable TDM mode I (20660) Adev_Codec: Open codec device OK I (20670) ES7210: Unmuted I (20670) Adev_Codec: Open codec device OK I (20670) AudioCodec: Set input enable to true I (20670) AudioCodec: Set output enable to true ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4200bbb6 --- 0x4200bbb6: BoxAudioCodec::Read(short*, int) at D:/esp32/OpenSource/jianglian-s3cam/xiaozhi-esp32-main/main/audio/codecs/box_audio_codec.cc:232 (discriminator 1) --- (inlined by) BoxAudioCodec::Read(short*, int) at D:/esp32/OpenSource/jianglian-s3cam/xiaozhi-esp32-main/main/audio/codecs/box_audio_codec.cc:230 (discriminator 1)
> 
> file: "./main/audio/codecs/box_audio_codec.cc" line 232 func: virtual int BoxAudioCodec::Read(int16_t*, int) expression: esp_codec_dev_read(input_dev_, (void*)dest, samples * sizeof(int16_t)) I (21670) AudioCodec: Set input enable to false E (21690) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet I (21700) I2S_IF: Mode 1 need extend bits 32 to 64 I (21700) I2S_IF: channel mode 2 bits:16/16 channel:4 mask:3 I (21710) I2S_IF: TDM Mode 0 bits:16/16 channel:4 sample_rate:24000 mask:3 I (21710) I2S_IF: channel mode 0 bits:16/32 channel:2 mask:1 I (21720) I2S_IF: STD Mode 1 bits:16/32 channel:2 sample_rate:24000 mask:1 I (21720) ES7210: Bits 16 I (21730) ES7210: Enable ES7210_INPUT_MIC1 I (21740) ES7210: Enable ES7210_INPUT_MIC2 I (21740) ES7210: Enable ES7210_INPUT_MIC3 I (21740) ES7210: Enable ES7210_INPUT_MIC4 I (21740) ES7210: Enable TDM mode I (21750) ES7210: Unmuted I (21750) Adev_Codec: Open codec device OK I (21750) AudioCodec: Set input enable to true I (29560) SystemInfo: free sram: 96303 minimal sram: 84175 I (36680) I2S_IF: Pending out channel for in channel running I (36680) AudioCodec: Set output enable to false I (39560) SystemInfo: free sram: 96279 minimal sram: 84175

78, 2025-08-13:
> 这个问题有新进展吗？我这边还是不断复现

最新版本是 1.8.7，这个问题已经在前面的版本已经修复过了

78, 2025-08-14:
> [@78](https://github.com/78) 虾哥可以试试一个corner case，网络正常连接，但是服务器返回的语音服务器url连接不上（你可以手动改成一个无效的）。这个问题基本上还是100%复现，即使在最新版本
> 
> ```
> E (53095) WebSocket: Failed to connect to server
> E (53095) WS: Failed to connect to websocket server
> I (53095) Application: STATE: idle
> E (53095) i2s_common: i2s_channel_disable(1218): the channel has not been enabled yet
> W (53095) Application: Alert Error: Unable to connect to service, please try again later [sad]
> I (53105) I2S_IF: channel mode 2 bits:16/16 channel:4 mask:3
> I (53115) I2S_IF: TDM Mode 0 bits:16/16 channel:4 sample_rate:24000 mask:3
> E (53125) i2s_common: i2s_channel_disable(1218): the channel has not been enabled yet
> I (53135) I2S_IF: channel mode 0 bits:16/16 channel:4 mask:3
> E (53135) i2s_common: i2s_alloc_dma_desc(510): allocate DMA buffer failed
> E (53145) i2s_std: i2s_std_set_slot(122): allocate memory for dma descriptor failed
> E (53155) i2s_std: i2s_channel_reconfig_std_slot(347): set i2s standard slot failed
> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x4205c031  PS      : 0x00060430  A0      : 0x8205c2bb  A1      : 0x3fcdc610  
> --- 0x4205c031: set_drv_fs at /home/weihengni/Projects/quechat-esp32/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:140
> A2      : 0x3c431f44  A3      : 0x00000001  A4      : 0x00000020  A5      : 0x3fcdaf1c  
> A6      : 0x00000000  A7      : 0x00000020  A8      : 0x00000000  A9      : 0x3fcdc5e0  
> A10     : 0x00000101  A11     : 0x3fcdc648  A12     : 0x3c1edfd8  A13     : 0x0000cf8f  
> A14     : 0x3fcd68d4  A15     : 0x3c431f44  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
> EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  
> --- 0x400570e8: memset in ROM
> --- 0x400570f3: memset in ROM
> ```

没有测出你说的问题

whni, 2025-08-14:
@78 我的硬件没有变化，就是但是从这个commit开始出现：https://github.com/78/xiaozhi-esp32/commit/6bb95073a41d77f795d74823f2c978463e4ba2e7

非常稳定复现，网络连接正常，但是语音服务器连接失败

whni, 2025-09-03:
依旧有这个问题，给一个不能连接的voice server，重复连接失败，有一定概率触发的

感觉是个race condition @78

78, 2025-09-03:
你只有一个设备么，换一个设备测试一下

whni, 2025-09-03:
我试过5个设备都有这个问题

whni, 2025-09-03:
是的，用过我们自己做的板子，也用过立创开发版。要把服务器地址改了，比如改成一个无法连接的url，这样就容易触发了

whni, 2025-09-04:
这个帖子不是我的帖子，所以其实算是有别的用户发现了这个问题😂