78/xiaozhi-esp32 #994: 刷入固件后黑屏错误不显示

vuong43i41, 2025-07-28:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

我在淘宝买了一个使用ESP32-S3-WROOM-1 MCN16R8芯片的机器人，它有一个1.4英寸的TFT显示屏。我刷入了在小智上下载的1.8.2固件后，重启后，机器人会提示“进入网络连接模式”，但屏幕黑屏，不显示任何内容。请问这是屏幕与新固件版本不兼容的问题吗？请帮我解决一下，非常感谢！

### What is the actual behavior?

刷入固件后黑屏错误不显示

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

w4mxl, 2026-01-04:
@vuong43i41 我买的也是 ESP32-S3-N16R8 带1.54寸屏幕的那款。在刷了征辰科技1.54TFT(wifi版)  v2.1.0版的固件后，出现了和你同样的黑屏情况，配网和语音交互都还是正常能用的，只是屏幕无法显示内容。请问你后面有找到解决这个问题的办法吗

go-maple, 2026-02-19:
很奇怪。官方固件可以，但是编译代码确实出问题了什么原因呢