78/xiaozhi-esp32 #1008: 摄像头参数询问

TTT-OS, 2025-07-30:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

立创实战派s3，版本1.7.6以上

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Power Supply used.

USB

### What is the expected behavior?

想了解一下是因为设备端重新统一化管理为NetworkInterface导致的还是

### What is the actual behavior?

[image]
这个是1.8.2版本，使用最新版的开源服务端（后面使用了自己重新写的视觉接口也是不行，发现是过不了aiohttp的校验，设定了Transfer-Encoding: chunked模式，但是有Content-Length的返回）

[image]
1.7.6版本可正常使用，因为还使用的旧Http* CreateHttp()的定义

### Steps to reproduce.

使用设备调用摄像头时就触发了

### Debug Logs.

```plain

```

78, 2025-07-30:
这个接口是只有 client 使用 Transfer-Encoding，server 不能使用 Transfer-Encoding。

TTT-OS, 2025-07-31:
> 这个接口是只有 client 使用 ，server 不能使用 的 header。`Transfer-Encoding: chunked``Transfer-Encoding: chunked`

打扰您一下，服务端没有设定Transfer-Encoding: chunked来着只负责接收响应，好像是接收到的响应里面Transfer-Encoding: chunked 但是又携带了Content-Length：0。
然后两者因HTTP/1.1 规范同一个请求中不能同时存在 Content-Length 和 Transfer-Encoding: chunked 头部导致的报错。
您指的是服务端错误的使用了带有Transfer-Encoding的header，才产生的相关报错吗？

78, 2025-07-31:
> 打扰您一下，服务端没有设定Transfer-Encoding: chunked来着只负责接收响应，好像是接收到的响应里面Transfer-Encoding: chunked 但是又携带了Content-Length：0。 然后两者因HTTP/1.1 规范同一个请求中不能同时存在 Content-Length 和 Transfer-Encoding: chunked 头部导致的报错。 [...]

我是看到你提供的图片里有 HTTP chunked response 的返回

TTT-OS, 2025-07-31:
> 我是看到你提供的图片里有 HTTP chunked response 的返回

那个是在1.7.6版本通过抓包设备端响应的header信息，然后上面是1.7.7以上的报错信息说设备端传了个双设定的冲突

78, 2025-07-31:
OK. 我似乎发现了问题，

request_chunked_ 的赋值应该放在 BuildHttpRequest 之前，你可以先手动调整一下位置。