78/xiaozhi-esp32 #1025: Unable to register from Reunion Island

j-kwan, 2025-08-01:
Hi,

I have tried to register with a Reunion Island number but don’t get a verification code.
Is there an alternative way to register or to receive the verification code please ?

Thanks

j-kwan, 2025-08-06:
Hello,
Thank you for your reply.
But how can I choose to receive the verification code with WhatsApp?
Is there an option I haven't found?
I just have the input fields for the phone number, the captcha and the verification code.

[image]

78, 2025-08-07:
> Hello, Thank you for your reply. But how can I choose to receive the verification code with WhatsApp? Is there an option I missed? I only have the input fields for the phone number, the captcha, and the verification code.
> 
> <img alt="Image" width="507" height="436" src="https://private-user-images.githubusercontent.com/1317359/475154804-a2a22608-fd88-453b-89dc-cddca57ddd5d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ1NDIzOTcsIm5iZiI6MTc1NDU0MjA5NywicGF0aCI6Ii8xMzE3MzU5LzQ3NTE1NDgwNC1hMmEyMjYwOC1mZDg4LTQ1M2ItODlkYy1jZGRjYTU3ZGRkNWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDgwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA4MDdUMDQ0ODE3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWMzOGIxYTEwOTJhNjkzZGVjYzgwYzQxYmIwNDEzZWZlMTg4MTEyZGQwZjI1NjkyMjkyM2MyN2VjZTZhNTE2MiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.SAk-jYtq1gN-jgMbgOrhZnOOynm4xAjZ7_fkPQpvDwk">

I think the verification code will be sent to you via WhatsApp. Please check your WhatsApp messages for the code to complete your registration.

j-kwan, 2025-08-11:
I tried several times and checked Whatsapp and my messages but I did not received the verification code :(

j-kwan, 2025-08-11:
Ok
Thank you very much.
I have sent an email.
The Google account login would be a great alternative too !

closed 2025-08-12 by j-kwan