78/xiaozhi-esp32 #1040: Disconnect display from spoken text

snooopy3652, 2025-08-06:
### Is your feature request related to a problem?

Hello,
I'm no expert or even have many skills in programming.
I got 3 different ESP32S3 devices running Xiaozhi and realized a problem that with the units with bigger displays.
If the spoken text is shown on the big display and the device "speaks" it will always stop for a moment when reaching the end of the shown text and continue after the screen changes.
So it is more or less "reading the display text".

The unit with the small screen does have endless scrolling text and because of this it also will not stop speaking for a second.
One solution would be to enable a scroll option on every device or don't "read" the screen but internaly also have a "scrolling" text.

78, 2025-08-07:
It shouldn't be like that.
What does your device look like?

snooopy3652, 2025-08-08:
It is the 1.28 box (round display no touch). 
Maybe it is me imagining that it is connected to the text on the display? I do use German language while using it and the model is yet not designed for speaking German so it sometimes does stumble but I think that this are different "problems"