78/xiaozhi-esp32 #1044: release.py脚本编译问题

NicLam26, 2025-08-08:
您好，

目前我用python .\scripts\release.py xmini-c3 这个命令编译的时候发现一个问题。

第一步：在SDK config 里面把OTA的地址改成自己电脑部署的服务器地址
第二步：用python .\scripts\release.py xmini-c3编译（config.json里面没有配置OTA的地址）
第三步：编译之后，发现OTA的地址会被覆盖成小智服务器的地址

请问这个脚本执行了什么，会强制覆盖服务器地址吗，看了好几遍编译脚本的代码也没有发现问题。

望解答，谢谢！

78, 2025-08-14:
> 具体的配置是这样写的：
> 
> <img alt="Image" width="1442" height="690" src="https://private-user-images.githubusercontent.com/218184781/477899106-1c5af350-0fdd-48a6-9f59-b1782cae6535.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTUxNTk1MzcsIm5iZiI6MTc1NTE1OTIzNywicGF0aCI6Ii8yMTgxODQ3ODEvNDc3ODk5MTA2LTFjNWFmMzUwLTBmZGQtNDhhNi05ZjU5LWIxNzgyY2FlNjUzNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgxNFQwODEzNTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMDkzMDIzYmRhYTAyNmMwMjVhYzI1OThkMjNlYjU0NTNkYzYyMWYyOTdlZmE3NTI2Mzc4NzhiNjU3NTFjOWY4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.KWLcZMrSgrf0kBYtEVAeXJuxbSGMfel-HJ9CEXteWe8">

字符串内容缺少双引号。 =\"http://.../\""

78, 2025-08-14:
按照我说的，或者你看 main/boards/atoms3r-echo-base/config.json 的写法

closed 2025-08-16 by NicLam26